OpenIntelWireless/itlwm #392: Can't build the kexts on Xcode 12 on Catalina

racka98, 2020-11-14:
**OS: Catalina 10.15.7**
**Kext version: 1.2.0**
**Xcode version: 12**

I can't build the kexts, I get errors on both Xcode and on command line using the command; `xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release`

## I get the following errors:

/Users/racka/Downloads/itlwm-master/itlwm/PrivateSPI.pch:1:10: fatal error: 
    'libkern/libkern.h' file not found
#include <libkern/libkern.h>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.

CompileC /Users/racka/Downloads/itlwm-master/build/itlwm.build/Release/itlwm.build/Objects-normal/x86_64/itlwm_info.o /Users/racka/Downloads/itlwm-master/build/itlwm.build/Release/itlwm.build/DerivedSources/itlwm_info.c normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'itlwm' from project 'itlwm')

**&**

/Users/racka/Downloads/itlwm-master/build/itlwm.build/Release/itlwm.build/DerivedSources/itlwm_info.c:1:10: fatal error: 
      'mach/mach_types.h' file not found
#include <mach/mach_types.h>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.

** BUILD FAILED **

The following build commands failed:
	ProcessPCH++ /Users/racka/Downloads/itlwm-master/build/SharedPrecompiledHeaders/SharedPrecompiledHeaders/6046404859436984703/PrivateSPI.pch.gch /Users/racka/Downloads/itlwm-master/itlwm/PrivateSPI.pch normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
	CompileC /Users/racka/Downloads/itlwm-master/build/itlwm.build/Release/itlwm.build/Objects-normal/x86_64/itlwm_info.o /Users/racka/Downloads/itlwm-master/build/itlwm.build/Release/itlwm.build/DerivedSources/itlwm_info.c normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
(2 failures)

williambj1, 2020-11-14:
Use MacKernelSDK

closed 2020-11-14 by williambj1

racka98, 2020-11-14:
> Use MacKernelSDK
Thanks!