OpenIntelWireless/itlwm #394: AirportItlwm detecting networks but not connecting

Freddson, 2020-11-14:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm 1.1.0 Stable
- WiFi Card Model: Intel Wireless Dualband N 7260
- Product ID: 0x08B1
- macOS Version: macOS 11 Big Sur

**Description**
After installing the Big Sur version of AirportItlwm, I tried force-loading IO80211Family and using AirportItlwm with my WiFi card. Although, even if the built-in menu finds networks, it cannot connect to them. I tried replacing the kext and rebuilding network properties, but none of these options worked. 

**Bug Report Archive**
[Report_2020-11-14_22-04-02.log](https://github.com/OpenIntelWireless/itlwm/files/5541460/Report_2020-11-14_22-04-02.log)

**Kext Download Source**
I downloaded the kext from the Github Releases tab, from the fast China URL.

win1010525, 2020-11-15:
Your log doesn’t contain itlwm’s log. Big Sur doesn’t have a full log.

Freddson, 2020-11-15:
> Your log doesn’t contain itlwm’s log. Big Sur doesn’t have a full log.

I did do what the troubleshooting guide told me to, what can I do to get it then?

williambj1, 2020-11-15:
@Freddson 

macOS Big Sur no longer shows logs from kexts injected by bootloaders. Please use [this method](https://openintelwireless.github.io/itlwm/Troubleshooting.html#getting-kext-logs-for-loading-failures) (Getting Kext Logs For Loading Failures) as a workaround and make sure you generate the logs as soon as possible after booting into the system. I will update the script later.

Freddson, 2020-11-15:
> @Freddson
>
> macOS Big Sur no longer shows logs from kexts injected by bootloaders. Please use [this [...]

Thank you, I will try to do it right now.

samwzlim, 2020-11-15:
Same issue here, airportitlwm was working flawlessly on Catalina but ever since I upgraded to big sur, it has not been working consistently, sometimes connected and sometimes not. On a side note, I have already updated my Keats folder with the airportitlwm kext for big sur

williambj1, 2020-11-15:
@samwzlim Can't tell what's wrong without logs.

Freddson, 2020-11-15:
I finally got the log! Sorry for making you wait. @williambj1 
[Log_2020-11-15-16-38-$S.log](https://github.com/OpenIntelWireless/itlwm/files/5542641/Log_2020-11-15-16-38-.S.log)

zxystd, 2020-11-16:
> I finally got the log! Sorry for making you wait. @williambj1
> [Log_2020-11-15-16-38-$S.log](https://github.com/OpenIntelWireless/itlwm/files/5542641/Log_2020-11-15-16-38-.S.log)

Nothing wrong according to the log.

Freddson, 2020-11-16:
> > I finally got the log! Sorry for making you wait. @williambj1
> > [Log_2020-11-15-16-38-$S.log](https://github.com/OpenIntelWireless/itlwm/files/5542641/Log_2020-11-15-16-38-.S.log)
> 
> Nothing wrong according to the log.

Though, I still can't connect. I tried redoing the whole EFI in an act of desperation and it still doesn't want to connect.

zxystd, 2020-11-16:
@Freddson @samwzlim can you describe more detail about the connecting behaviour? now i can not find anything related to your information, also, maybe you can collect more logs(when the problem happen again). thanks.

Freddson, 2020-11-16:
Sure, though I might have to reinstall itlwm. Now no networks show up at all, even though WiFi is turned on.

zxystd, 2020-11-18:
@Freddson don't need to reinstall, just turn off and then turn on the wifi, if it is already loaded.

Freddson, 2020-11-18:
Alright

Freddson, 2020-11-18:
It happened again, I got a new log and it has a "getSSIDData" error.

Freddson, 2020-11-18:
[Log_2020-11-18-13-18-08.log](https://github.com/OpenIntelWireless/itlwm/files/5559952/Log_2020-11-18-13-18-08.log)
[Log_2020-11-18-13-18-24.log](https://github.com/OpenIntelWireless/itlwm/files/5559953/Log_2020-11-18-13-18-24.log)
Two logs from itlwm and AirportItlwm.

midi1996, 2020-11-18:
I don't know if this is the same issue, but I'm having the same behavior, I have a 8265 card.
[Report_2020-11-18_22-09-25.log](https://github.com/OpenIntelWireless/itlwm/files/5562781/Report_2020-11-18_22-09-25.log)
When connecting to 2.4Ghz networks it works fine, however it wont connect to 5Ghz network from my router and says that the passcode is wrong (which is not). 
I tried wifi hotspot from my phone (5Ghz, as seen by my other laptop with a broadcom card) and my intel setup still isn't accepting it.
I have a 8260 that I can try, but will it really make a change?

midi1996, 2020-11-19:
@williambj1 
ok so here is the setup:
- Intel Wireless 8265
- macOS 11.0.1
Wireless APs tested:
- A 2.4Ghz network from a Huawei router/modem (40Mhz, CH6) - AP1
- A 2.4Ghz network from a TPLINK router with OpenWRT (40Mhz, CH10) - AP2 
- A 5Ghz network from the same TPLINK (80Mhz, CH40) - AP3 
- A 5Ghz network tethering AP from phone (Galaxy S7) (80Mhz, CH40) - AP4

Test with kexts
- itlwm 1.1.0 stable + heliport stable
 - ✅ AP1
 - ✅ AP2
 - 🚫 AP3 - no IP
 - 🚫 AP4 - no IP
- itlwm 1.2.0 + heliport stable
 - ✅ AP1
 - ✅ AP2
 - 🚫 AP3 - no IP
 - ✅ AP4 (I tried again with 1.1.0, and it wouldnt connect)
  - [Log_2020-11-19_01-17-57.log](https://github.com/OpenIntelWireless/itlwm/files/5563548/Log_2020-11-19_01-17-57.log)
- AirportItlwm 1.2.0 (log in the previous comment)
 - ✅ AP1
 - ✅ AP2
 - 🚫 AP3 - no IP
 - 🚫 AP4 - no IP
- AirportItlwm 1.0 (1.1.0)
 - ✅ AP1
 - ✅ AP2
 - 🚫 AP3 - no IP
 - 🚫 AP4 - no IP
  - [AirItlDebug_2020-11-19_01-28-30.zip](https://github.com/OpenIntelWireless/itlwm/files/5563556/AirItlDebug_2020-11-19_01-28-30.zip)

On another note, HeliPort's "diagnostics report" doesn't create logs from dmesg/system logs, just its own logs + some lines to check if itlwm is loaded, is this expected?

Freddson, 2020-11-19:
Yeah, williambj1 said earlier that bigsur broke diagnostic logs.

Freddson, 2020-11-19:
Alright, now the wifi totally stopped working. WiFi preferences say that it's always turned off, when I turn it on nothing happens. I'll reinstall AirportItlwm.

Freddson, 2020-11-20:
Redid the thing, troubleshooted again, networks show up again and still the same result.

williambj1, 2020-11-21:
@midi1996 Does AP3 & AP4 only support 80MHz connections? What happens if you force it to use 20MHz? Could you try Channel 36, 5GHz, 20MHz?

midi1996, 2020-11-21:
@williambj1 AP3 is my router so I can change it to 20Mhz, but that still didn't work. (I confirmed those changes on my other laptop and my phone to see if the channels changed to 36 (20Mhz))
[AirItlDebug_2020-11-21_20-32-53.zip](https://github.com/OpenIntelWireless/itlwm/files/5578263/AirItlDebug_2020-11-21_20-32-53.zip) (this is with AP3 @ those settings you suggested)
This and other issues started piling up:

- WiFi menu saying that my 2.4Ghz network is hidden (which is not)
- Switching between 2.4Ghz and 5Ghz networks breaks everything (cannot connect to either of them until I reboot)
- It always says "Network not found" when switching between frequencies (but if it's between 2.4Ghz networks, it works fine, only if I never touch 5Ghz)
- When switching, it says that said network has "WEP" instead of whatever it was using and then stops from connecting (hard to reproduce, you have to switch *while* it's connecting to a network, at that point, it wont connect even if you waited).

Xpl0itU, 2020-11-21:
Sometimes that happens to me, go to settings, network, remove the wifi entry, apply it and add a new wifi entry, if that doesn’t fix it a reboot works 100% of the time

midi1996, 2020-11-21:
@Xpl0itU you mean with 5Ghz wifi entries?

midi1996, 2020-11-24:
This has been fixed with @zxystd (check gitter) (for my part of the issue)

Freddson, 2020-11-24:
Now that is great

Freddson, 2020-11-24:
I updated the kext, it now connects new networks which is good, but it still doesn't connect. Though, I got the logs and I saw an interesting thing. After itlwm connects to the network, it sends a setDEAUTH error and says that it Unhandles something. Here is the log:

[Log_2020-11-24-20-10-$s.log](https://github.com/OpenIntelWireless/itlwm/files/5592176/Log_2020-11-24-20-10-.s.log)

zxystd, 2020-11-24:
@Freddson check your passeord.

Freddson, 2020-11-25:
Already did, still the same issue.

Freddson, 2020-11-25:
I did some more troubleshooting on my network one more time, and it works! Thanks a lot for the help, i'll close this issue now.

closed 2020-11-25 by Freddson

Freddson, 2020-11-25:
Aaaaand it turns out there is a new error. The menu says it's connected to wifi but all apps say it's offline.

_**pain.**_

It is fixable by turning wifi off and on via a usb dongle menu I installed. Still, it isn't what I want. I got the logs again and they're the same as the last time.

Freddson, 2020-11-26:
Turns out the problem only appears with new networks, the original network I connected to works perfectly.