OpenIntelWireless/itlwm #396: PHY mode is always 802.11n, even with 5GHz only network

grimfrost, 2020-11-15:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: AirportItlwm_v1.0_Beta_Catalina
- WiFi Card Model: Intel Dual Band Wireless-AC 7260
- Product ID: 7260.HMW
- macOS Version: macOS Catalina 10.15.7

**Description**
PHY mode is always 802.11n, even with 5GHz only network. The card and router I'm using supports 802.11ac.
[image]

**Bug Report Archive**
"Read this guide if you are using AirportItlwm http://localhost:8080/itlwm/Troubleshooting.html#manually-for-airportitlwm" -> ** This description gives me a localhost link. So I'm pretty unable to follow.**

**Kext Download Source**
Gitlab Releases

grimfrost, 2020-11-15:
Typo fix:
**Kext Download Source**
Github Releases

williambj1, 2020-11-15:
This is correct

closed 2020-11-15 by williambj1

grimfrost, 2020-11-15:
> This is correct

So I believe I missed the documented part where this issue mentioned?

williambj1, 2020-11-15:
Read the FAQ for more information.

grimfrost, 2020-11-15:
> Read the FAQ for more information.

Oh I just noticed "Performance Sucks". Thanks for your kind guidance.

aspenluxxxy, 2020-11-16:
> It is recommended for daily use  
> 
> ...  
> # Performance SUCKS

kinda contradicting, no?

also, is it just me, or is niche stuff like AirDrop being prioritized above actual features that matter like 802.11ac? If I knew how to help, I would, though.

grimfrost, 2020-11-16:
> > It is recommended for daily use
> > ...
> > # Performance SUCKS
> 
> kinda contradicting, no?
> 
> also, is it just me, or is niche stuff like AirDrop being prioritized above actual features that matter like 802.11ac? If I knew how to help, I would, though.

Well, Idk about you but that niche stuff is one of the reasons why I want to use Hackintosh instead of Windows. Of course I do appreciate performance but it is not most important point for me.

1Revenger1, 2020-11-16:
That stuff is being worked on in the background as far as I can tell - getting AMPDU/ac work is a huge task. I wouldn't be suprised if a lot of the AirDrop stuff is just from more stuff being reversed while working on ac (though I honestly have no clue and am not trying to speak for the team).

aspenluxxxy, 2020-11-16:
Oh, thanks for the explanation!