OpenIntelWireless/itlwm #397: [Airportitlwm 1.1] Airportitlwm stuck at loading itlwm on Big Sur

hoanghohuy, 2020-11-15:
**Have You Read Our Docs**
 Yes

**Are You Reporting A Bug**
 Yes

**Environment**

- Kext Version: AirportItlwm_v1.0_Beta_BigSur.kext
- WiFi Card Model: Intel AC-8265
- Product ID: 8086
- macOS Version: BigSur 11

**Description**
On Catalina it working
On BigSur, it stucked at loading itlwm on startup
Thanks.
Photo:
[image]

FaneCH, 2020-11-15:
I have the same issue on big sur. Can't extract any logs cause it doesn't boot.

win1010525, 2020-11-15:
Update to newest kext and try again.

FaneCH, 2020-11-15:
@win1010525 it does the same thing.
I also tried to disable ASPM but nothing happens.

win1010525, 2020-11-15:
Can you boot normally without itlwm?

williambj1, 2020-11-15:
Unrelated

closed 2020-11-15 by williambj1

FaneCH, 2020-11-15:
> Unrelated

Why the issue was closed?? This is a real bug that needs a fix!