OpenIntelWireless/itlwm #398: AX210适配问题

OctopusYuYu, 2020-11-16:
大佬您好 Intel的AX系列网卡出了AX210型号 
0x8086:0x2725
近期考虑适配吗？

williambj1, 2020-11-16:
Will but won't guarantee when.

zxystd, 2020-11-16:
卖太贵了，便宜了再说，现在用户也很少。

win1010525, 2020-11-16:
哈哈哈谁赞助z大一个

YBN-JUAN, 2020-11-16:
ax210价格是ax200的两倍就离谱，性能提升也不大。

win1010525, 2020-11-16:
而且linux貌似没找到固件

hezeniu, 2020-11-28:
> 卖太贵了，便宜了再说，现在用户也很少。

不少了Z大，这个卡配的蓝牙ID是您蓝牙驱动列表中的最后一个，这个省事了。抽空给支持了呗。在10.14.6上用xcode10.3编译itlwm,想尝试替换ID去支持 2725 ，没成功。估计是头文件老。Z大用的是哪个版本xcode 和 command tools 版本 能说下吗？

感谢Z大的大贡献。

williambj1, 2020-11-29:
> > 卖太贵了，便宜了再说，现在用户也很少。
> 
> 不少了Z大，这个卡配的蓝牙ID是您蓝牙驱动列表中的最后一个，这个省事了。抽空给支持了呗。在10.14.6上用xcode10.3编译itlwm,想尝试替换ID去支持 2725 ，没成功。估计是头文件老。Z大用的是哪个版本xcode 和 command tools 版本 能说下吗？
> 
> 感谢Z大的大贡献。

USB ID 是多少，蓝牙问题请在隔壁仓库开 issue。

zxystd, 2020-11-29:
> > 卖太贵了，便宜了再说，现在用户也很少。
> 
> 不少了Z大，这个卡配的蓝牙ID是您蓝牙驱动列表中的最后一个，这个省事了。抽空给支持了呗。在10.14.6上用xcode10.3编译itlwm,想尝试替换ID去支持 2725 ，没成功。估计是头文件老。Z大用的是哪个版本xcode 和 command tools 版本 能说下吗？
> 
> 感谢Z大的大贡献。

如果只是加id就能支持的话那反手就提交了，我何必要去买一个网卡呢？ax210是三代设备，具体想了解更多的话看文档吧。

hezeniu, 2020-11-29:
> > > 卖太贵了，便宜了再说，现在用户也很少。
> > 
> > 
> > 不少了Z大，这个卡配的蓝牙ID是您蓝牙驱动列表中的最后一个，这个省事了。抽空给支持了呗。在10.14.6上用xcode10.3编译itlwm,想尝试替换ID去支持 2725 ，没成功。估计是头文件老。Z大用的是哪个版本xcode 和 command tools 版本 能说下吗？
> > 感谢Z大的大贡献。
> 
> USB ID 是多少，蓝牙问题请在隔壁仓库开 issue。

隔壁已支持，哈哈。0x8087, 0x0032

mavisguo, 2020-12-10:
> > > > 卖太贵了，便宜了再说，现在用户也很少。
> > > 
> > > 
> > > 不少了Z大，这个卡配的蓝牙ID是您蓝牙驱动列表中的最后一个，这个省事了。抽空给支持了呗。在10.14.6上用xcode10.3编译itlwm,想尝试替换ID去支持 2725 ，没成功。估计是头文件老。Z大用的是哪个版本xcode 和 command tools 版本 能说下吗？
> > > 感谢Z大的大贡献。
> > 
> > 
> > USB ID 是多少，蓝牙问题请在隔壁仓库开 issue。
> 
> 隔壁已支持，哈哈。0x8087, 0x0032

所以现在AX210的蓝牙能用，而WIFI还要接着等吧？

KB5201314, 2020-12-11:
> 而且linux貌似没找到固件

试了下某宝买的AX210，firmware是有的: 
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/iwlwifi-ty-a0-gf-a0-59.ucode?id=7eb7fda50e9aa554c6bfafdd456e6c2ea54f6163

但是存在一个问题，subsystem device id 遗漏导致iwlwifi识别不了：
https://lore.kernel.org/linux-wireless/iwlwifi.20201202143859.a06ba7540449.I7390305d088a49c1043c9b489154fe057989c18f@changeid/T/#u
把这里的patch打上就好了↑

williambj1, 2020-12-11:
> 试了下某宝买的AX210，firmware是有的:
> https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/iwlwifi-ty-a0-gf-a0-59.ucode?id=7eb7fda50e9aa554c6bfafdd456e6c2ea54f6163 [...]

https://github.com/OpenIntelWireless/itlwm/issues/398#issuecomment-735348048

zxystd, 2020-12-11:
双十二来了，眼睁睁看着价格从79涨到了88. /摸鱼

williambj1, 2020-12-25:
Duplicate of #446

closed 2020-12-25 by williambj1