OpenIntelWireless/itlwm #400: Location Still ZZ in latest build(1.2.0 alpha)

seungjunProgramming, 2020-11-17:
OK
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: 1.2.0 alpha
- WiFi Card Model: AC8265
- Product ID: 0x24FD
- macOS Version: 11.0.1

**Description**

itlwmAirport's resent release shows up that It resolved the issue which prints ZZ in location .
But mine still remains ZZ.

Is it a known bug or do I have a problem???

**Bug Report Archive**

Not a Bug, but rather a question

**Kext Download Source**

build with the following command:

git clone --depth=1 https://github.com/OpenIntelWireless/itlwm.git
cd itlwm
git clone --depth=1 https://github.com/acidanthera/MacKernelSDK.git
git reset --hard HEAD
rm -rf build
git pull
rm include/FwBinary.cpp
find itlwm/firmware/ -type f ! -name 'iwm-8265-*' -delete
xcodebuild -project itlwm.xcodeproj -target fw_gen -configuration Release -sdk macosx
xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Big\ Sur -configuration Release -sdk macosx
echo "kext inside build/Release!!"
echo " "

seungjunProgramming, 2020-11-17:
Oops...Found the solution on the gitter
Sorry!

closed 2020-11-17 by seungjunProgramming

mehmetyukcell, 2020-11-17:
Which solution?