OpenIntelWireless/itlwm #402: Boot time on Big Sur really long

Yot360, 2020-11-17:
Hey, I updated to Big Sur and now my boot times went from 20s to 5mins (on ssd). I watched the boot process with verbose on, and like 2/3 of the boot process is itlwm, I think it’s really slowing down my boot speed but I need it for WiFi.

VoltaVX, 2020-11-17:
It's normal for ITLWM to spam your verbose mode a bit, but it doesn't prevent the system from booting up. I think you might be missing [the Catalina AHCI Kext](https://github.com/dortania/OpenCore-Install-Guide/blob/master/extra-files/CtlnaAHCIPort.kext.zip) which got removed in Big Sur. Add that kext to OpenCore, and your computer should be able to boot up as fast as before.

williambj1, 2020-11-17:
Duplicate of #353

closed 2020-11-17 by williambj1

Yot360, 2020-11-18:
> It's normal for ITLWM to spam your verbose mode a bit, but it doesn't prevent the system from booting up. I think you might be missing [the Catalina AHCI [...]

Thanks gonna try