OpenIntelWireless/itlwm #403: AC9462刚发布的1.2.0版本无法进入系统

hqting, 2020-11-18:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 
- WiFi Card Model: 
- Product ID: 
- macOS Version: 

**Description**

**Bug Report Archive**

**Kext Download Source**

chenyuyou, 2020-11-18:
我的也是。

Environment

Kext Version: V1.2.0-alpha
WiFi Card Model: AC9560；
Product ID: 0x8086
macOS Version: big sur 正式版
设备 id：0x34

williambj1, 2020-11-18:
Confirmed

williambj1, 2020-11-18:
[Panic-66f82e815a85920e6de61329881f5abeda3b7569.txt](https://github.com/OpenIntelWireless/itlwm/files/5559808/Panic-66f82e815a85920e6de61329881f5abeda3b7569.txt)

williambj1, 2020-11-18:
@chenyuyou @hqting 请提供更多信息，如 panic 日志。

zxystd, 2020-11-18:
please attach the panic log

hqting, 2020-11-18:
> [Panic-66f82e815a85920e6de61329881f5abeda3b7569.txt](https://github.com/OpenIntelWireless/itlwm/files/5559808/Panic-66f82e815a85920e6de61329881f5abeda3b7569.txt)

我想请教一下如何操作才能取得panic log这个文件。

williambj1, 2020-11-18:
Fixed in 69aaea39895ea7f78480315eab0cf4ac1f445250

closed 2020-11-18 by williambj1