OpenIntelWireless/itlwm #405: macOS 10.15.7 itlwm 或 HeliPort 无法正常连接 WiFi 不知是我系统问题还是驱动或 APP 问题？ 请教大佬。

godcheese, 2020-11-19:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: v1.1.0 stable
- WiFi Card Model: 
- Product ID: 
- macOS Version: 10.15.7

**Description**

**Bug Report Archive**

**Kext Download Source**

以下是我抓取的诊断报告：
[bugreport_27997.zip](https://github.com/OpenIntelWireless/itlwm/files/5564469/bugreport_27997.zip)

zxystd, 2020-11-19:
路由器配置问题

godcheese, 2020-11-19:
是否为网卡？网卡兼容问题么？以下图中一个是我买的 USB WiFi 网卡，一个是大佬的 HeliPort。
当我的 USB WiFi 网卡功能开启时，HeliPort 可用。但是当我的 USB WiFi 网卡功能关闭时 HeliPort 就不可用？即出现连接问题，无法正常联网。
[image]

[image]

godcheese, 2020-11-19:
> 路由器配置问题

按理说不会的，因为我用的是手机热点和公司 WiFi 都试过了，都是无法用 HeliPort 连接。之前我是 10.15.4 时，HeliPort 和 itlwm 正常驱动的，也是正常联网的，我前天才升级的系统，重装了下 HeliPort 和 itlwm 就出此问题了。

godcheese, 2020-11-19:
这个问题依据官方文档 [Resetting System Preferences - Network ](https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#resetting-system-preferences-network) 解决，不过 HeliPort 无法忘记 WiFi 的问题确实存在。望大佬能够解决。

zxystd, 2020-11-19:
你是不是从来没点开过"打开网络偏好设置"这个菜单

closed 2020-11-19 by zxystd

godcheese, 2020-11-19:
打开删除过2次。

Orgnge, 2020-11-19:
你可以试试自己把驱动通过OCC 直接打进去A