OpenIntelWireless/itlwm #406: itlwm+Heliport可以正常使用，使用AirportItlwm.kext无法开启Wi-Fi扫描网络。

HKMV, 2020-11-19:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes
**Environment**

- Kext Version: AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-3a2daed.zip
- WiFi Card Model: AC 8265
- Product ID: 0x24fd
- macOS Version: 10.15.7---11.0.1

**Description**
请问各位作者大佬，我使用itlwm+Heliport可以正常使用，但是使用AirportItlwm.kext在10.15.7和11.0.1都无法开启Wi-Fi扫描网络。如下图，请各位大佬明示：
[image]
[image]

CodeApePro, 2020-11-20:
按照发布文档,确认下面的内容
1.OC版本大于等于0.6.1
2.config里面设置SecureBootModel值为Default

HKMV, 2020-11-20:
> 按照发布文档,确认下面的内容
> 1.OC版本大于等于0.6.1
> 2.config里面设置SecureBootModel值为Default

1.我的OC版本是0.6.3
2.不是说macOS 版本 >= macOS11 就不用设为默认吗？

zxystd, 2020-11-20:
删除桥接器。重置网络设置。

closed 2020-11-20 by zxystd

HKMV, 2020-11-20:
> 删除桥接器。重置网络设置。

成功的解决了我的问题，感谢大佬🧍‍♂️

zyx121, 2023-06-11:
> > 删除桥接器。重置网络设置。
> 
> 成功的解决了我的问题，感谢大佬standing_man

哥，我系统版本venture，我也是之前airport，用着就是定位不行。后来先删除了kexts里的itwm,接着把config里面的itwm的√取消 然后在这两个地方分别吧air portltlwm.kext放进去。然后打开就WiFi感叹号，我的oc是 0.8几，那个smb也设置defeat了。这是怎么回事啊哥。是我操作有问题吗？

zyx121, 2023-06-11:
> > 删除桥接器。重置网络设置。
> 
> 成功的解决了我的问题，感谢大佬standing_man

有可能您不会回我了，时间跨度太大了

HKMV, 2023-06-12:
> > > 删除桥接器。重置网络设置。
> > 
> > 成功的解决了我的问题，感谢大佬standing_man
> 
> 有可能您不会回我了，时间跨度太大了

我现在一直使用的AirportItlwm.kext，没有发现你这样情况，我系统版本是13.4,OC是0.9.1