OpenIntelWireless/itlwm #412: Sidecar seems to work , tx rate not accurate

samleong123, 2020-11-22:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No. Only Info

**Environment**

- Kext Version: v1.1.0
- WiFi Card Model: Intel AX201
- Product ID: 8086,2f0
- macOS Version: 10.15.7

**Description**
Sidecar seems to be able to detect but unable to connect. Handoff and Universal Clipboard worked perfectly! Airdrop doesn't work.
[image]
[image]
[image]
[image]

And i found out that the tx rate is 140-150Mbps but even internal speed test is reaching only 50Mbps on 5GHz , kind of weird.
The speedtest server is connected through 1GBPS Cat 5e to my router with using IIS and i use WiFi 5GHz to connect it 

**Bug Report Archive**

**Kext Download Source**
Your Repo

samleong123, 2020-11-22:
#### Latest Info : Sidecar seems to be work and connected to my iPad!
[image]
[image]

williambj1, 2020-11-22:
Tx rate is correct, please read our docs for more info. Sidecar is not supposed to work, please connect with a cable instead.

closed 2020-11-22 by williambj1