OpenIntelWireless/itlwm #413: Cannot "Turn Wi-Fi On" with AirportItlwm

arrmo, 2020-11-23:
**Have You Read Our Docs**
Yes! 😄

**Are You Reporting A Bug**
I believe so - only saying that as I won't argue it may be me in the end. Always possible!

**Environment**
- Kext Version: Have tried AirportItlwm and itlwm (more on that below), v1.1 and 1.2 (to test).
- WiFi Card Model: Cudy AX3000 (AX200)
- Product ID: Network controller [0280]: Intel Corporation Wi-Fi 6 AX200 [8086:2723] (rev 1a)
- macOS Version: 11.0.1

**Description**
I admit, part of this is me, figuring out the "correct" options / combinations. If I have it right (yell if not!),
a) itlwm works with HeliPort => does seem to work, have to configure Wi-Fi through HeliPort, and then in macOS network connections it shows up as Ethernet (right?).
b) AirportItlwm does not work with HeliPort, it says itlwm is not loaded (makes sense). Rather, macOS network connections shows the connection as Wi-Fi (perfect, what I want!). But my issue is that the device is Off, and trying to "Turn Wi-Fi On" (button) doesn't seem to do anything. I see the same thing with v1.1 or v1.2.

**Bug Report Archive**

a) Not using Heliport ... OK?

b) Can't seem to get to this page - perhaps because it says WiFi is off?

**Kext Download Source**

Downloaded directly from your site, And to the instructions, I followed the second option / step for AirportItlwm, meaning ```Force IO80211Family to load. 「Supports OpenCore and Clover(not tested)」 (Read the Kernel - Force section in OpenCore's manual for more info)```. I checked, and do get the following,
```
~ % sudo kextstat | grep -i "itlwm"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   68    0 0                  0xd46000   0xd46000   com.zxystd.AirportItlwm (1.2.0) B8084376-4F13-3794-B5ED-E644E1B3BE32 <67 46 14 6 5 3 1>
~ % sudo kextstat | grep -i "IO80211"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   67    1 0xffffff8001c3a000 0x13a000   0x13a000   com.apple.iokit.IO80211Family (1200.12.2b1) EBD11055-A4E6-3243-A60C-0AAAD85C12E3 <66 64 46 28 10 8 7 6 5 3 1>
```
So it does seem that AirportItlwm and IO80211Family are loaded, right?

Thoughts? Things I can provide that would help?

Thanks!

williambj1, 2020-11-23:
Please attach logs extracted by this method: https://openintelwireless.github.io/itlwm/Troubleshooting.html#using-dmesg

> \# Using dmesg

arrmo, 2020-11-23:
You bet! Attached. Just yell if you need me to try anything.

Thanks!
[Log_2020-11-22_19-05-46.log](https://github.com/OpenIntelWireless/itlwm/files/5580749/Log_2020-11-22_19-05-46.log)

arrmo, 2020-11-23:
FYI, tried the instructions outlined here,
https://macpaw.com/how-to/fix-wi-fi-problems-on-mac#:~:text=Delete%20and%20reset%20Wi%2DFi%20network%20settings&text=Or%20move%20them%20to%20a%20new%20folder%20on%20your%20Desktop.&text=Now%20you%20will%20have%20to,your%20Wi%2DFi%20network%20again.

After the step "Delete and reset Wi-Fi network settings" => now no WiFi device showing up. As part of the install, perhaps some sort of config needs to be done?

Thanks!

arrmo, 2020-11-24:
Hmmm ... OK, this may be me, but ... 😞. Tried stepping back to itlwm (from AirportItlwm), see if I get my network interfaces back. Heliport says that itlwm is not loaded, but I confirmed,
```
https://openintelwireless.github.io/itlwm/Troubleshooting.html#kernel-extension-loading-status
```

Any suggestions of how to get the network interface "back"? I do see that it's missing (```ifconfig```). Thanks!

williambj1, 2020-11-24:
Have you tried resetting your network interface configs?

arrmo, 2020-11-24:
LOL - funny, as I just did that, reboot just completed 😄. Yep, all network interfaces recreated - except this one.

FYI, I don't seem to have the files (after reboot),
```
com.apple.network.identification.plist
com.apple.wifi.message-tracer.plist
```

BTW, I also deleted ```com.apple.network.eapolclient.configuration.plist``` ... should that not be done?

Thanks!

arrmo, 2020-11-24:
OK, got it back ... sort of 🤣. Did a full clean (network reset), power cycle, reset NVRAM => device is back. Seems to come up with itlwm + HeliPort, but of course shows as an Ethernet connection in Network Preferences. So then, switched to AirportItlwm, but still now showing as Ethernet (of course then, can't control it from macOS WiFi settings).

Thoughts?

Thanks!

arrmo, 2020-11-24:
OK, pulling my hair out - and not enough of it to spare ... LOL. I got back to working, as above, but moved to AirportItlwm (not working, as noted), so went back to itlwm ... and it's not working again either (HeliPort complaining, can't turn WiFi on). Hmmm ... something is not getting set it seems (and yes, rebooted in each case). I also tried (all with no luck),
1) Shutdown and restart
2) NVRAM reset
3) Manual kextunload, load

Other things to try?

Thanks!

arrmo, 2020-11-24:
OK, kinda got it working - let me explain the "cycle", may help you, and at least you know about it going forward 😄.

Making any changes, even a version of the same driver (e.g. AirportItlwm) => I have to fully shut down, restart. Even a reboot with the same driver, no joy ... always requires a full power down. Seems like something is not being reset quite right perhaps?

Thanks again for all the help!

Rixxx-git, 2020-11-25:
I am having the exact same issues!

Can you please let me know:

1. How exactly did you reset network configs? Which configs should I delete and which not?
2. How do I reset NVRAM and is it safe with the latest OpenCore Version?
3. Do you also use the IntelBluetoothFirmware kexts?

arrmo, 2020-11-25:
Hi. To your questions,

1. I'm not ... that can be done (as per the docs), but doesn't really seem to be a fix / workaround.
2. Yes, have tried this too - and SEEMS safe 🤣. But also, not a fix.
3. Nope

Sort of glad I'm not the only person seeing this, but it is odd. A full host OS reboot, just to reboot the guest OS? Seems strange. And FYI, when things come back up (incorrectly) ... I still see the network interface, but it's showing as Ethernet, not Wi-Fi. Do you see the same?

Perhaps some sort of race condition / sequencing issue?

Thanks!

Rixxx-git, 2020-11-27:
Thanks for your help! 

For me it was all running fine but then I did a BIOS update and some other changes to the EFI folder and now I am at the stage where it doesn't show any networks. Not with itlwm and heliport or with airportitlwm.

How did you fix that?

Before I did the changes wifi was working with itlwm and heliport but still shown as Ethernet. 🙂

arrmo, 2020-11-27:
> Before I did the changes wifi was working with itlwm and heliport but still shown as Ethernet

Yes, I would expect that - the same as I saw. If you use itlwm, it seems to be recognized as Ethernet. If you use AirportItlwm, then it will be Wi-Fi, and you need to configure with built-in / macOS controls.

Rixxx-git, 2020-11-27:
Ok, thank you! But how do I get it back to work? At the moment both itlwm and airportitlwm can't find any network. What did you do about that problem?

arrmo, 2020-11-27:
From your comments above, it seems that your BIOS and/or EFI changes caused this to stop working - right? Can you undo those?

Rixxx-git, 2020-11-27:
No I can't rollback the BIOS. It works for other users with the same BIOS version, so it might be something else. How did you fix your problem? I might just try the same.

arrmo, 2020-11-27:
I just made sure to only enable AirportItlwm (and NOT itlwm), then a full host reboot (not just guest). Does that not work for you?

Rixxx-git, 2020-11-27:
no it doesnt 😔 can you check my EFI folder for mistakes?

arrmo, 2020-11-27:
EFI folder? Not sure where that is.

You are force-loading 80211, right?

Rixxx-git, 2020-11-27:
what is 80211?

arrmo, 2020-11-27:
Meaning the step outlined here,
https://openintelwireless.github.io/itlwm/Installation.html#airportitlwm

williambj1, 2020-11-28:
Please use gitter or other sources for personal support. This is clearly not a bug.

closed 2020-11-28 by williambj1

arrmo, 2020-11-28:
Hi,

Yes, the latest sidetrack isn't about a bug - agreed! But I'm not so sure that the issue I initially raised isn't. I cannot get the driver to work without a full host OS reboot. That doesn't seem right to me (i.e. something isn't getting configured right during boot it seems).

Thoughts?

Thanks!

Rixxx-git, 2020-11-29:
I still can't turn on Heliport. Here are my log files:
[HeliPort_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5612123/HeliPort_logs.log)
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5612124/itlwm_logs.log)

I did a reset of the Network settings like described (https://openintelwireless.github.io/itlwm/Troubleshooting.html#resetting-system-preferences-network) but nothing changes. Any ideas?

arrmo, 2020-11-29:
Sorry, nothing else I can think of. And this (real bug?) got closed, as it sidetracked into setup debugging 😞.

Rixxx-git, 2020-11-29:
😔

labrouss, 2020-11-29:
What is your device id ;

Kind regards,
Konstantinos