OpenIntelWireless/itlwm #417: Intel® Dual Band Wireless-AC 3168 Wifi connection issue

borasumer, 2020-11-24:
**Have You Read Our Docs**
Yes
**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 
Latest
- WiFi Card Model: 
Intel® Dual Band Wireless-AC 3168
- Product ID: 
- macOS Version: 

**Description**
When I turn the pc on, first 10 mins internet is not working and wifi is not connected. I try to connect the wifi, it does, but there is no connection. I turn wifi on and off couple of times and it starts working. Not sure this is a common behaviour, if not then that could be a bug for Big Sur.

**Bug Report Archive**

Can't go this address ?

**Kext Download Source**
This repo

zxystd, 2020-11-24:
attach log please.

PriNcEFuL, 2020-11-25:
I had same problem but using latest alpha version fix the problem... now it happens rarely once a week but fixes it self after a min or 2

borasumer, 2020-11-25:
> I had same problem but using latest alpha version fix the problem... now it happens rarely once a week but fixes it self after a min or 2

Good to hear, I can update the kext with the alpha one then, thanks a bunch mate.

borasumer, 2020-11-26:
Yes alpha version/v1.2.0-alpha solved my issue. Closing the issue.

closed 2020-11-26 by borasumer