OpenIntelWireless/itlwm #418: Bluetooth 5

henkiewie, 2020-11-24:
Hi,
To my surprise I got WiFi working on my ryzentosh with the latest airportitwlm.kext with my intel x200 chipset on Catalina 10.15.7 on opencore 0.6.4.  The heliport app isn't working (doesn't seem to find any BT, but I'm just using the standard WiFi options. I have to reconnect by hand after reboot (adding my ssid and password in the info.plist did not seem to work).  Bluetooth was already working before using the new kext, but it seems to be stuck on an older or unkonown version.
I'm looking for a way to force the version to Bluetooth 5 mainly because of the delay when using audio. But I have no idea how. I have a headphone Sony that supports bt5 and aptx but changing the parameters doesn't work it's SBC only for now. I can channge this when I connect the headphone to my android phone it's a hassle, but just to let you know the headphone supports this. Also turning on aac does nothing. When I look for the lmp version it states unknown 0xa. I've been using bluetooth explorer and the command line to try make it happen, but it doesn't.
I tried the bluetoothfirmware.kext, but it seems to break my bluetooth.
Some technical info;
kext;
72 0 0xffffff7f84180000 0xa24000  0xa24000  com.zxystd.AirportItlwm (1) D4493512-24B9-3B2C-89CE-98C65E75A04F <71 34 13 6 5 3 1> 
opencore; 
REL-064-2020-11-22opencore;
Bluetooth;
Apple Bluetooth Software Version: 7.0.6f7
Hardware, Features and Settings:
Name: iMac Pro
Address: AC-67-5D-A3-E9-53
Bluetooth Low Energy Supported: Yes
Handoff Supported: Yes
Instant Hot Spot Supported: Yes
Manufacturer: Intel
Transport: USB
Firmware Version: 10.256 (10.256)
Bluetooth Power: On
Discoverable: Off
Connectable: Yes
Auto Seek Pointing: On
Remote wake: On
Vendor ID: 0x8087
Product ID: 0x0029
Bluetooth Core Spec: Unknown (0xA)
HCI Revision: 0x100
LMP Version: Unknown (0xA)
LMP Subversion: 0x100
Device Type (Major): Computer
Device Type (Complete): Mac Desktop
Composite Class Of Device: 0x380104
Device Class (Major): 0x01
Device Class (Minor): 0x01
Service Class: 0x1C0
Auto Seek Keyboard: On
Devices (Paired, Configured, etc.):
WH-1000XM3:
Address: CC-98-8B-3E-46-37
Major Type: Audio
Minor Type: Headset
Services: Wireless iAP, Hands-Free unit, Amazon Alexa, AudioWear, Headset, CSR GAIA‚Ñ¢, AudioWear Voice
Paired: Yes
Configured: Yes
Connected: No
Manufacturer: Cambridge Silicon Radio (0x7, 0x3907)
Bluetooth Core Spec: 4,1
Firmware Version: 0x0452
Vendor ID: 0x054C
Product ID: 0x0CD3
Class of Device: 0x04 0x01 0x240404
EDR Supported: Yes
eSCO Supported: Yes
SSP Supported: Yes
Services:
Bluetooth File Transfer:
Folder other devices can browse: ~/Public
When receiving items: Accept all without warning
State: Disabled
Bluetooth File Exchange:
Folder for accepted items: ~/Downloads
When other items are accepted: Save to location
When receiving items: Accept all without warning
State: Disabled
Bluetooth Internet Sharing:
State: Disabled
Incoming Serial Ports:
Bluetooth-Incoming-Port:
RFCOMM Channel: 3
Requires Authentication: No

zxystd, 2020-11-24:
macOS didn't support bt5.1 and 5.2, so it shows as unknown. also, itlwm just a wifi related kext, not bluetooth, please use intelbluetoothfirmware.

closed 2020-11-24 by zxystd

henkiewie, 2020-11-24:
Thanks for the info and your work on the WIFI kext.. Will try troubleshooting the Bluetooth kext. 
Regards,
Henk