OpenIntelWireless/itlwm #419: Connection constantly dropping/inable to access internet while being connected

ScrapyHD, 2020-11-25:
Hi, i stay connected to my wifi network but i can only access the internet for 30 seconds before it bugs out and says that i have no connection, im still connected to the wifi network but it acts as if the wifi has no internet connection. When i disable and re-enable the wifi it works again for a webpage or 2 then stops again. I have attached the log, any help is appreciated thanks!
[bugreport_56630.zip](https://github.com/OpenIntelWireless/itlwm/files/5594353/bugreport_56630.zip)

zxystd, 2020-11-27:
Router compatibility problem, will not be fix in short. I will open a detail issue about this and mark `help wanted`. Now the only solution is to change to another router. Thanks.

ScrapyHD, 2020-11-27:
ok will do, thanks for the help!

grimfrost, 2020-11-27:
This also happens on my system too. 
My router is ZTE H267A & Wifi card is AC7260. 
Just ordered another router (not because of this problem. It was in my mind for a while). 
It is a Keenetic Extra, See if that new one changes anything.

zxystd, 2020-11-30:
@grimfrost how about the new router?

lgariv, 2020-12-05:
Happened to me as well for the last 3 weeks or so,  even tho I haven't changed my router. Last alpha version seems to have fixed it (although it might be a different problem or not related).

erenatas, 2020-12-10:
Not to complain, but to report. I am also having this problem with my ISP modem. It is very very weird though. I was on a GoToMeeting event, the connection on it was ok, but I was not able to send any HTTP or SSH requests from any client. Some sort of transport layer problem I guess.

dlinsenm, 2020-12-28:
I too am having this issue on my AX200. I am not having this issue in Windows on this machine. It works for about several minutes, then I have to turn it off and on again.

Kuibagit, 2020-12-29:
网卡AX201也有此现象，使用2~5分钟就会断流，不管是连路由器热点还是iPhone/Android热点都会出样现象。
---------
System：MacOS Catalina 10.15.7
HeliPort：1.0.1
itlwm：1.1.0

zxystd, 2021-01-01:
can you guys try to test itlwm-1.2.0-aplha version? maybe Tx-aggregation will improve the compatibility.

ahalsaqqa, 2021-01-01:
Same issue here on latest alpha release

dlinsenm, 2021-01-01:
> can you guys try to test itlwm-1.2.0-aplha version? maybe Tx-aggregation will improve the compatibility.

I am now on the alpha but am still experiencing the same issue. I'm attaching the logs in case this is helpful. Thanks for your work!
[bugreport_12550.zip](https://github.com/OpenIntelWireless/itlwm/files/5759348/bugreport_12550.zip)

leonurdo, 2021-01-31:
I am experiencing the same issue. i can access the internet 15-20 min than lost internet connection I am  still connected to the wifi network..

MrCreeper1008, 2021-01-31:
A workaround for Android users is to use USB tethering via your phone by loading this kext: https://github.com/jwise/HoRNDIS

GiacomoPasc, 2021-02-07:
I have the same issue, but with the last beta [dbdbf00] it seems to lost connection after 30 seconds, I tried other versions but I can't have a stable connection [Max 30 min.].

My network card is Intel Ac3160. I Hope this will help. Anyway thanks for the amazing work.

Kuibagit, 2021-02-08:
> I have the same issue, but with the last beta [dbdbf00] it seems to lost connection after 30 seconds, I tried other versions but I can't have a stable connection [Max 30 min.].
> I tried with android hotspot and it keeps connected forever.
> 
> My network card is Intel Ac3160 and I run Catalina 10.15.7 . I Hope this will help. Anyway thanks for the amazing work.
> 
> 
> 

Setring the MTU size worked for me.
you can try  https://github.com/OpenIntelWireless/itlwm/issues/477#issuecomment-753438847

GiacomoPasc, 2021-02-09:
> > I have the same issue, but with the last beta [[dbdbf00](https://github.com/OpenIntelWireless/itlwm/commit/dbdbf007ac5bc102cd3d487b3084f5949a383978)] it seems to lost connection after 30 seconds, I tried other versions but I can't have a stable connection [Max 30 min.].
> > I tried with android hotspot and it keeps connected forever.
> > My network card is Intel Ac3160 and I run Catalina 10.15.7 . I Hope this will help. Anyway thanks for the amazing work.
> 
> Setring the MTU size worked for me.
> you can try [#477 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/477#issuecomment-753438847)

It seems to work now. I set the MTU size to 1464. I own an Italian Vodafone Router. Thank you very much for the advice.

lgariv, 2021-02-25:
Intel AC-9462 here, macOS Catalina, having the same problem since the latest alpha, here's some insight:

1. Itlwm has a more stable connection than AirportItlwm for some reason.
2. I can not connect to 2.4GHz access points with both Itlwm and AirportItlwm.
3. Before the latest alpha, I could connect to a new access point with WPA3 enabled (in WPA2/WPA3 mode), but now it is not working on both Itlwm and AirportItlwm.
4. I can not maintain a good connection / speed (I purposely refer to both, I'm not confusing them) unless I'm in the next room or in the same room with the router. By that I mean that I either get connection drops or just that the Wi-Fi speeds just slow down over time - I couldn't find out what situation is causing either to happen (to me).
5. I get about 95/70mbps with Itlwm while only getting 20/15mbps with AirportItlwm.

I will edit this comment to include logs as soon as I'm back to using macOS.

win1010525, 2021-02-25:
> Intel AC-9462 here, macOS Catalina, having the same problem since the latest alpha, here's some insight:
> 
> 1. Itlwm has a more stable connection than AirportItlwm for some reason.
> 2. I can not connect to 2.4GHz access points with both Itlwm and AirportItlwm.
> 3. Before the latest alpha, I could connect to a new access point with WPA3 enabled (in WPA2/WPA3 mode), but now it is not working on both Itlwm and AirportItlwm.
> 4. I can not maintain a good connection / speed (I purposely refer to both, I'm not confusing them) unless I'm in the next room or in the same room with the router. By that I mean that I either get connection drops or just that the Wi-Fi speeds just slow down over time - I couldn't find out what situation is causing either to happen (to me).
> 5. I get about 95/70mbps with Itlwm while only getting 20/15mbps with AirportItlwm.
> 
> I should also mention that my router is TP-Link AX20; I have tried to set it manually to work on 20MHz, 40 MHz, 80MHz and Auto modes and they all behave the same with latest alpha.
> MTU is set on both the router and network configurations to 1492.
> 
> I will edit this comment to include logs as soon as I'm back to using macOS.

Airportitlwm have a more serious speed issue because of the constantly scanning.

lgariv, 2021-02-25:
> Airportitlwm have a more serious speed issue because of the constantly scanning.

In the previous version I had about the same speed in both of them tho

hectorox, 2021-03-15:
I have the same issue on the latest release. Constantly signal dropping and/or disconnection from internet.

zxystd, 2021-07-24:
@lgariv 
> 2. I can not connect to 2.4GHz access points with both Itlwm and AirportItlwm.

The problem is fixed in the latest commit 375d8a514b7772133e233135ab636a458e69447c I think.

williambj1, 2021-10-02:
@ScrapyHD Does the same behavior still occur in the latest release?

williambj1, 2022-01-18:
Assuming fixed

closed 2022-01-18 by williambj1