OpenIntelWireless/itlwm #423: Wireless-AC 9560 [Jefferson Peak] 无线网卡无法正常工作

snowy2004, 2020-11-27:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: all
- WiFi Card Model: Wireless-AC 9560 [Jefferson Peak] 
- Product ID: 0xA370
- macOS Version: 10.15.7

**Description**
itlwm加载日志里提示：kernel: (itlwm) itlwm: itlwm: failed to read nvm

日志文件及设备信息如下：
[Log_2020-11-28_07-08-24.log](https://github.com/OpenIntelWireless/itlwm/files/5608544/Log_2020-11-28_07-08-24.log)
[pcidevices.txt](https://github.com/OpenIntelWireless/itlwm/files/5608547/pcidevices.txt)

win1010525, 2020-11-28:
所有版本都不行？

snowy2004, 2020-11-28:
@win1010525 就是提示 itlwm: failed to read nvm 难道这个网卡有问题吗

williambj1, 2020-11-29:
Remove extra PCI device properties in your config.

closed 2020-11-29 by williambj1