OpenIntelWireless/itlwm #428: Wi-Fi comes and leaves

TheFailureOne, 2020-11-29:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: 
1.10 if I'm not wrong
- WiFi Card Model: 
AC 3168
- Product ID: 
ES1-572-36XH (if this is what you mean with product ID
- macOS Version: 
MacOS 11.0.1 Big Sur 
**Description**
So, ever since updating from Catalina to Big Sur I've been having Wi-Fi problems where it basically works for a few minutes and then it stops working and  viceversa.
Also, this bug occurs when I select the Wi-Fi trough the Info.plist file, if I change it trough Heliport it straight up doesn't work, it just gives me an exclamation mark, please help
**Bug Report Archive**
I can't get them right now since it's giving me the problem, once I can I will make sure to send you the archive
**Kext Download Source**
I got it from the GitHub page and it's no custom built

williambj1, 2020-11-29:
Please obtain logs using this method. https://openintelwireless.github.io/itlwm/Troubleshooting.html#using-dmesg

TheFailureOne, 2020-11-29:
> Please obtain logs using this method. https://openintelwireless.github.io/itlwm/Troubleshooting.html#using-dmesg

[bugreport_25554.zip](https://github.com/OpenIntelWireless/itlwm/files/5612759/bugreport_25554.zip)
[bugreport_16513.zip](https://github.com/OpenIntelWireless/itlwm/files/5612760/bugreport_16513.zip)

williambj1, 2020-11-30:
You should use `dmesg` to obtain logs. Will reopen once the process is done correctly.

closed 2020-11-30 by williambj1

TheFailureOne, 2020-11-30:
> You should use `dmesg` to obtain logs. Will reopen once the process is done correctly.

[Log_2020-11-30_17-26-54.log](https://github.com/OpenIntelWireless/itlwm/files/5617197/Log_2020-11-30_17-26-54.log)
here you go

TheFailureOne, 2020-11-30:
> You should use `dmesg` to obtain logs. Will reopen once the process is done correctly.

Here it is
[Log_2020-11-30_17-31-46.log](https://github.com/OpenIntelWireless/itlwm/files/5617321/Log_2020-11-30_17-31-46.log)
(The one before was wrong, it still didn't disconnect to the internet)

TheFailureOne, 2020-12-01:
> You should use `dmesg` to obtain logs. Will reopen once the process is done correctly.

Are you going to reopen it?

TheFailureOne, 2020-12-02:
Ok, I think I fixed it, I just reinstalled Big Sur