OpenIntelWireless/itlwm #429: Heliport: "Turn on Wi-Fi" doesn't work

Rixxx-git, 2020-11-29:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: itlwm 1.1.0 & Heliport 1.0.1.
- WiFi Card Model: Intel Wireless 8265 / 8275
- Product ID: HP 830 G5 - OpenCore 0.63
- macOS Version: 11.0.1.

**Description**
Heliport: "Turn on Wi-Fi" doesn't work
I did a Network Preference reset like described in Troubleshooting
I did a NVRAM reset
I cleared and rebuilt Kext Cache

**Bug Report Archive**
[bugreport_52973.zip](https://github.com/OpenIntelWireless/itlwm/files/5612182/bugreport_52973.zip)

**Kext Download Source**
Downloaded the kext here: https://github.com/OpenIntelWireless/itlwm/releases

williambj1, 2020-11-29:
> For me it was all running fine but then I did a BIOS update and some other changes to the EFI folder and now I am at the stage where it doesn't show any networks. Not with itlwm and heliport or with airportitlwm.

If it was working for you before, it is clearly not our issue, please check your hardware/bios/configs.

closed 2020-11-29 by williambj1

labrouss, 2020-11-29:
With same device Id, I’ve got Airportitlwm working. Try Alpha version.

Kind regards,
Konstantinos

Rixxx-git, 2020-11-29:
> With same device Id, I’ve got Airportitlwm working. Try Alpha version. Kind regards, Konstantinos
> […](#)

I tried that already. Which BIOS version do you use?