OpenIntelWireless/itlwm #431: Kernel Panic when waking from sleep

FaneCH, 2020-11-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: v1.2.0-alpha (AirportItlwm
- WiFi Card Model: Intel AC-3165
- Product ID: 0x3166
- macOS Version: 10.15.7

**Description**
When waking my laptop from sleep, i get a black screen and after rebooting, on the KP log, itlwm is mentioned (see the attached image).

**Bug Report Archive**
[Archive.zip](https://github.com/OpenIntelWireless/itlwm/files/5614941/Archive.zip)
[image]

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.2.0-alpha

zxystd, 2020-11-30:
EFI related issue. Thanks for the feedback.

FaneCH, 2020-11-30:
No it's not ... if i disable your kext, nothing happens.

rockavoldy, 2020-12-06:
i get this KP too when wakeup from sleep, i don't think this is related to my EFI config. last commit i tested that don't have this KP is [07eb9cf](https://github.com/OpenIntelWireless/itlwm/commit/07eb9cf2b7fc8c8e6aaffe4eebed2bc8f6d8ff05).

from commit [d9bd8d4](https://github.com/OpenIntelWireless/itlwm/commit/d9bd8d435b80c009c928156ef805e14b2a527fd7) to latest trigger this KP randomly when wakeup from sleep.

**Environment**
Kext Version: v1.2.0-alpha (AirportItlwm)
WiFi Card Model: Intel AC-3168
Product ID: 0x24fb
macOS Version: 10.15.7 (19H15)

poohzaza166, 2020-12-29:
what is your laptop model since some will need a acpi to fix sleep problem

closed 2021-01-16 by williambj1