OpenIntelWireless/itlwm #433: AirportItlwm stops working after sleep/wake - BigSur - AC 9260

hrytskivr, 2020-11-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: AirportItlwm_v1.0_Beta_BigSur.kext
- WiFi Card Model: AC 9260
- Product ID: vendor: 0x8086, device: 0x2526
- macOS Version: 11.0.1

**Description**
AirportItlwm.kext works fine until I put my laptop to sleep. After wake it will take long time to connect to network, and even if it succeeds in connecting, it will say that there is no internet connection and won't work, after restart everything works again.

**Bug Report Archive**
[AirItlDebug_2020-11-30_11-21-49.zip](https://github.com/OpenIntelWireless/itlwm/files/5614983/AirItlDebug_2020-11-30_11-21-49.zip)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.1.0

zxystd, 2020-11-30:
Try pressing `DHCP` button in wifi network panel when awake. Not sure if it is a bug on Big Sur because someone had told me that real Mac also have this problem.

hrytskivr, 2020-11-30:
@zxystd I don't see such button, only Renew DHCP Lease in advanced/tcpip. I haven't had this issue when using Fenvi BCM94360NG.

hrytskivr, 2020-11-30:
@zxystd I've just noticed same behaviour on my ax200 on desktop computer. Big Sur also, same kexts.

teabyii, 2020-12-03:
I also encountered the same problem. 
AX200, macOS 11.0.1, AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-c2f2c51.zip

meaverick, 2020-12-08:
I also encountered the same problem.
AX200, macOS 11.0.1,

ThePonyZ, 2020-12-09:
我发觉尝试连接Wi-Fi后可以看到PHY模式先是802.11s，然后断开，重连换成802.11g，又断开，然后再连802.11n才能获取到IP，不知道是否与这个有关？

dvrkoo, 2020-12-16:
I'm also having this issue on Big Sur on an intel ac 7260

Litarvan, 2020-12-17:
I'm also having this issue on two completely different computers, one with Intel 9560, I don't know about the other one. But it seems that "Renew DHCP" sometimes fixes the issue after waiting for a bit, thanks @zxystd

dvrkoo, 2020-12-18:
airportilwm 1.2.0 fixed the issue for me

eric2877, 2020-12-25:
airporttwlwm 1.2.0 debug fix that problem for me too，but it still not stable, loss connection sometime in daily usage, I have to reconnect by hand.

hrytskivr, 2020-12-25:
airportilwm 1.2.0 debug fixed that in a way for me, but when waking with that version wifi is altogether not working until manually off/on, then I can connect.

wqhqq1, 2021-01-29:
I have the same problem on 1.3.0 ax200

wqhqq1, 2021-02-01:
It's fixed now or you may have to check your USB settings in Hackintool.

williambj1, 2021-10-02:
Feel free to post comments with log files if the same behavior still occurs in the latest release.

closed 2021-10-02 by williambj1

KenDxD, 2022-07-03:
I currently experiencing this issue on my IntelAC3165 wifi card on monterey, after waking my laptop from sleep the wifi is not working. checking it on terminal it says inactive