OpenIntelWireless/itlwm #434: AirportItlwm stopped working suddenly, wifi is not working anymore

Le0xFF, 2020-11-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm v1.2.0-alpha
- WiFi Card Model: Intel AX200
- Product ID: 0x2723
- macOS Version: 11.0.1 (but it happened also in 10.15.7)

**Description**
Wi-Fi was working fine, no problem at all.

After changin ROM (en0 MAC Address) in my config.plist, my wifi stopped working at all. Now I have a wifi icon in my top bar but no Wi-Fi interface in SystemSettings -> Network and I cannot add any wifi interface at all. Bluetooth is working fine tho.
Maybe I made some mistake with en0?

I also fresh installed BigSur with a fresh OpenCore folder but no luck at all. I also tried resetting NVRAM but nothing.
Maybe is something related to my BIOS settings? But I changed nothing at all about my wifi setting there.

**Bug Report Archive**
I attached my log created with msgbuf.
[AirportItlwmLog.txt](https://github.com/OpenIntelWireless/itlwm/files/5619061/AirportItlwmLog.txt)

**Kext Download Source**
[AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-aa2f1a7.zip](https://github.com/OpenIntelWireless/itlwm/releases/download/v1.2.0-alpha/AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-aa2f1a7.zip)

Le0xFF, 2020-11-30:
... After a lot of reboots, wifi was showing up again in SystemSettings -> Network, without any other changes to my PC.
What the hell.

closed 2020-11-30 by Le0xFF

micabad, 2021-07-07:
I also experienced this. After booting to recovery and enable SIP: ">csrutil enable", everything works fine. Bluetooth and WiFI work well. BTW I'm using Intel AC3168