OpenIntelWireless/itlwm #440: Slow connection when using Microsoft Remote Desktop

Creedowl, 2020-12-03:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: itlwm: 1.2.0 beta fw: 34.0.1
- WiFi Card Model: Intel AC 8265
- Product ID: 0x24fd
- macOS Version: 11.0.1

**Description**

The connection is very slow when I connect to windows remote desktop using Microsoft Remote Desktop, whether in LAN or WAN, with high latency and very low resolution. Instead when I connect via wired network the rdp works well.

**Bug Report Archive**

[AirItlDebug_2020-12-03_19-40-58.zip](https://github.com/OpenIntelWireless/itlwm/files/5635869/AirItlDebug_2020-12-03_19-40-58.zip)

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.2.0-alpha

williambj1, 2020-12-03:
Assuming complaining about performance. Closed.

closed 2020-12-03 by williambj1

Creedowl, 2020-12-03:
@williambj1 I'm not complaining about performance, just wandering why this issue occured, because rdp does't require high  bandwidth. I guess it may relate to udp issue