OpenIntelWireless/itlwm #446: [Feature] ax210 and other 3rd generation cards support.

zxystd, 2020-12-11:
1. Buy a ax210 card.
2. Update all of the firmwares to new version and do lots of messy firmware compatibility work.
3. add gen3 supported codes according to linux iwlwifi.

BTW, I don't think it will take a high priority.

qqsir-dev, 2020-12-16:
AX210 end user really need this work!!!! great news!!

hsimi, 2020-12-23:
感谢大大

thefiredragon, 2021-01-30:
I would buy one for you and me @zxystd

ConnHowe, 2021-04-25:
We are wating for you!

zxystd, 2021-05-04:
ax210 card is becoming more and more expensive...

win1010525, 2021-05-04:
By giving zxystd AX210, it may accelerate the developing process. QwQ.🌝🙏

thefiredragon, 2021-05-04:
Ich would donate for Hardware, so theres a way for donarions?

Dave1482, 2021-05-27:
https://www.ebay.com/itm/193733803381?chn=ps&mkevt=1&mkcid=28

this seems pretty reasonable, looking forward to WiFi 6E support lol

wqhqq1, 2021-06-11:
It still doesn't work

wqhqq1, 2021-06-11:
Is there any plans?

williambj1, 2021-06-11:
@wqhqq1 Feel free to open a pull request and submit a patch.

wqhqq1, 2021-06-11:
@williambj1 Is there any document about writing wifi driver? I've learnt c++ before, maybe could help a bit.

wqhqq1, 2021-06-12:
Trying force enable but got this:
[    1.445952]: AirportItlwm::init AirPortFamily_kexts-1850.7 "AirPortFamily_kexts-1850.7" May  8 2021 05:06:20
[    1.477338]: itlwm: : Why??? unknown adapter type
[    1.477340]: itlwm: ieee80211_ifdetach
[    1.477342]: itlwm: ieee80211_proto_detach
[    1.477343]: itlwm: ieee80211_crypto_detach
[    1.477345]: itlwm: ieee80211_node_detach
[    1.477347]: itlwm: ifmedia_delete_instance
[    1.477348]: itlwm: attach fail
[    1.477906]: itlwm: free
[    1.477907]: itlwm: virtual void ItlHalService::free()
[image]

wqhqq1, 2021-06-27:
Is there any news?

zxystd, 2021-07-18:
[image]
Hello everyone, good news. ax210 is work now, and I will merge these changes in next release version(v2.1.0 maybe) since the code is changed a lot.

seungjunProgramming, 2021-07-18:
> [image]
> Hello everyone, good news. ax210 is work now, and I will merge these changes in next release version(v2.1.0 maybe) since the code is changed a lot.

Will there be any feature implements to the kext at the 2.1.0 version???
(For ex. Airdrop maybe??!!)

Dave1482, 2021-07-27:
> [image]
> Hello everyone, good news. ax210 is work now, and I will merge these changes in next release version(v2.1.0 maybe) since the code is changed a lot.

would you happen to have a guess when it will be released? I'm really excited for this! Well done btw

Chase-Tsien, 2021-08-06:
Looking forward to the new version.I almost replace my AX210 with AX200 or other cards.Thanks the zxystd for the effort.

Fu1haha, 2021-08-13:
为什么我使用了ax210以后会经常导致系统卡死，大佬们需要什么日志文件来研究下吗 我小白一个 不是很懂

williambj1, 2021-08-13:
> 为什么我使用了ax210以后会经常导致系统卡死，大佬们需要什么日志文件来研究下吗 我小白一个 不是很懂

The DMA allocation issue hasn’t been solved yet.

Fu1haha, 2021-08-13:
> > 为什么我使用了ax210以后会经常导致系统卡死，大佬们需要什么日志文件来研究下吗 我小白一个 不是很懂
> 
> The DMA allocation issue hasn’t been solved yet.

幸苦了！

GilDev, 2022-01-13:
[“Compatibility” page of the OpenIntelWireless documentation](https://openintelwireless.github.io/itlwm/Compat.html) still says AX210 is uncompatible.