OpenIntelWireless/itlwm #452: Turn this into a sponsored project?

bolu-atx, 2020-12-13:
Thanks a lot for this effort, I am forever grateful, have you guys considered turning this into a sponsored project? I think a lot of us would like to express our appreciation through a token of gratitude.

跪谢各位大神，让我的电脑可以跑 hacknitosh来调试mac端程序

lunjielee, 2020-12-15:
这样不会被Apple DMCA吗？毕竟这里面有许多Reverse Engineering的东西

borasumer, 2020-12-16:
> Thanks a lot for this effort, I am forever grateful, have you guys considered turning this into a sponsored project? I think a lot of us would like to express our appreciation through a token of gratitude.
> 
> 跪谢各位大神，让我的电脑可以跑 hacknitosh来调试mac端程序

I don't think they can accept money for this.

zxystd, 2020-12-18:
Thank you for the advice, we definitely need help and support, but we don't have any sponsor plan. the reason was described in the doc. Anyone can help us in others way, eg: code contribution, feature testing and report, help to answer/check new issue and so on.

closed 2020-12-18 by williambj1