OpenIntelWireless/itlwm #455: Airportitlwm just stops working after macOS was in sleep mode for a long time

GGorAA, 2020-12-16:
**Have You Read Our Docs**
True

**Are You Reporting A Bug**
True
**Environment**

- Kext Version: Latest (1.2.0)
- WiFi Card Model: 8260
- Product ID: idk what it is, but here are the screenshots from Hackintool:
[image]
[image]

- macOS Version: Big Sur 11.0.1 (20B29)

**Description**

Airportitlwm probably crashes when macOS goes to sleep mode for 15 minutes and longer. It is pretty annoying, because I probably need to restart the whole Hackintosh to make this thing work aggain

richardkindom, 2020-12-17:
Currently do not sleep if plugged into an AC. This works for me for most cases.

williambj1, 2020-12-17:
You have unsupported device properties that enable ASPM, please remove it.

closed 2020-12-17 by williambj1