OpenIntelWireless/itlwm #459: Unable to connect using Intel Wireless AC-9260

morning-sucks, 2020-12-20:
I'm trying to use AirportItlwm.kext with my build (has an integrated Intel Wireless AC 9260), but it simply refuses to work. However, my screen will not display at all (but still is operating apparently, my resolution changes to 1080p when I type in my password and press enter, and I can reboot the system by opening a terminal blind and typing in sudo reboot and my password) if AirportItlwm.kext is not present within /EFI/OC/Kexts and in the OpenCore config.plist. I've rebooted my system around 30 times, and the Wi-Fi has actually worked exactly once, and I was able to use my system like a normal Mac for the most part, but it stopped, and that was the only time it ever happened. It works in recovery mode just fine too, btw. Bluetooth works no matter what I do, which is weird.

I've tried everything I know up till loading AirportItlwm.kext manually from Terminal (because of the screen problem I mentioned), extracting the IO80211Family.kext from /System/Library/Extensions (I don't know how to do it), and disabling SIP and putting the kext in /Library/Extensions. Here is what I've done so far:

1. Loaded it normally using OpenCore with SecureBootModel set to Default.
2. Forced IO80211Family.kext to load from OpenCore, along with coreupdate.kext
3. Replaced with itlwm.kext and installed HeliPort.
4. Removed all preferences for network and rebooted.

With the first two, it boots normally and takes me to the login screen, but the Wi-Fi either is stuck with the bars moving up and down, or says "The network cannot be connected to." The third one simply doesn't allow me to even see a screen. The fourth one does nothing.

I'm at an impasse, so here are my specs; I don't think everything here is even relevant but I'm providing it anyway:

CPU: AMD Ryzen 9 3950X  
Cooler: Scythe Mugen 5  
Motherboard: ASUS TUF GAMING X570-PLUS (WI-FI)  
RAM: G.SKILL Trident RGB DDR4-3600 32GB  
Storage: Sabrent Rocket 4.0 1 TB  
Graphics: ASRock CHALLENGER D OC RX 5500 XT 8 GB  
Ethernet: Realtek L8200A (not going to use it since it's unsupported)  
Wi-Fi/Bluetooth: Built-in Intel Wireless AC 9260  

Here are the kexts I'm using:

- AirportItlwm.kext
- AppleALC.kext
- AppleMCEReporterDisabler.kext
- Lilu.kext
- NVMeFix.kext
- VirtaulSMC.kext
- WhateverGreen.kext

Appreciate all the help I can get, thanks in advance.

williambj1, 2020-12-20:
Please use dmesg to generate logs and post it here, instructions can be found in our docs.

Also, please keep in mind that we did not test this kext with AMD MBs, there might be some vendor incompatibility issues.

williambj1, 2020-12-25:
Feel free to continue commenting. Closed due to inactivity.

closed 2020-12-25 by williambj1