OpenIntelWireless/itlwm #463: itlwm is not running

raagk, 2020-12-22:
Hey I have intel AC 8265.
I tried to use itlwm.kext with Heliport and get itlwm is not running after opening Heliport.
Please note that he kext is loaded as seen in terminal 
[image]

Also, i tried to take up logs,
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5727799/itlwm.log)
[Log_2.log](https://github.com/OpenIntelWireless/itlwm/files/5727800/Log_2.log) (heliport).

Please tell me if I am doing something wrong, I am using OpenCore for kext injection.

raagk, 2020-12-22:
Here is dimes log 
[Log_2020-12-22_10-25-10.log](https://github.com/OpenIntelWireless/itlwm/files/5728020/Log_2020-12-22_10-25-10.log)

zxystd, 2020-12-22:
There is no log in your attachment and 8265 is confirmed to work. please fix your system/EFI first. I think it is not a driver issue. Thanks. BTW, can you provide your card's product id?

raagk, 2020-12-22:
Hey! Thanks for taking up my issue.
Product ID:	0x8001
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.03
  Speed:	Up to 480 Mb/s
  Location ID:	0x1d100000 / 1
  Current Available (mA):	500
  Current Required (mA):	0
  Extra Operating Current (mA):	0

raagk, 2020-12-22:
Also, can you tell what possibly could cause this issue in EFI? I have checked drivers and config sanity checker. It seems fine!

williambj1, 2020-12-25:
Sorry but EFI related issues are off topic. Please use other sources for personal help.

closed 2020-12-25 by williambj1

raagk, 2020-12-25:
@williambj1 Is the Product ID or Vendor ID correct? My WiFi module is detected as USB Hub.