OpenIntelWireless/itlwm #467: Intel Centrino N2230 support

hect-dpr, 2020-12-24:
Hey guys great work, but is the intel centrino n2230 from Lenovo thinkpad edge E431 supported? The kext seems to have loaded, but Heliport says itlwm is not running.

Thanks

win1010525, 2020-12-25:
Take a look at doc. Not tested card.

pigworlds, 2020-12-25:
Please collect and attach logs.
What's the PCI ID of this card? Given this is a wireless card released long time ago, it may not be supported.

hect-dpr, 2020-12-25:
Here are the logs and PCI IDS. Thanks
[bugreport_35690.zip](https://github.com/OpenIntelWireless/itlwm/files/5741380/bugreport_35690.zip)
[pcidevices.txt](https://github.com/OpenIntelWireless/itlwm/files/5741382/pcidevices.txt)

pigworlds, 2020-12-25:
The logs is incomplete or the driver is not loaded properly.
Can you follow https://openintelwireless.github.io/itlwm/Troubleshooting.html#getting-kext-logs-for-loading-failures ?

hect-dpr, 2020-12-26:
Here are those logs mate.
[Log_2020-12-26_16-55-21_dmesg.log](https://github.com/OpenIntelWireless/itlwm/files/5743037/Log_2020-12-26_16-55-21_dmesg.log)
[Log_2020-12-26_17-09-06_manual.log](https://github.com/OpenIntelWireless/itlwm/files/5743039/Log_2020-12-26_17-09-06_manual.log)

pigworlds, 2020-12-26:
The logs indicate you're using v1.1.0.

The experimental support for Centrino cards is added after v1.1.0.

Please try latest v1.2.0-alpha instead.

hect-dpr, 2020-12-27:
OMG, it started working after loading v1.2.0-alpha mate. Thank you very much. I never thought Intel WiFi would work in a hackintosh. Kudos!

closed 2020-12-27 by hect-dpr