OpenIntelWireless/itlwm #468: AX200通过heilport能够搜索并连接但是无网络

VioletLing, 2020-12-25:
[image]
安装好itlwm与heilport之后，能够正常搜索到wifi，也能够正常链接，在路由器内部能够看到设备已成功链接。但是Wi-Fi标志会显示为一个叹号（图中是因为使用了USB共享网络，然后Wi-Fi标志为正常），网络设备中也没有显示有新的设备，无法进行网络连接。
After installing itlwm and heilport, the wifi can be searched normally, and the connection can be normal. You can see that the device has been successfully connected inside the router. However, the Wi-Fi logo will be displayed as an exclamation mark (the figure is because the USB shared network is used, and the Wi-Fi logo is normal), and there is no new device displayed in the network device, and the network connection cannot be made.

VioletLing, 2020-12-25:
[system.log](https://github.com/OpenIntelWireless/itlwm/files/5742273/system.log)

closed 2020-12-26 by VioletLing

jiaxin96, 2021-01-02:
问题解决了吗？