OpenIntelWireless/itlwm #469: Bluetooth stops functioning when wifi enabled

Xodiced, 2020-12-27:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
YES

**Environment**
- Kext Version: 1.2.0
- WiFi Card Model: Intel Wireless-AC 7265 + Bluetooth 4.2
- Product ID: 
- macOS Version: 11.1

**Description**
When WiFi is enabled with the AirPort driver Bluetooth stops functioning.  The minute you toggle WiFi off Bluetooth devices work instantly again.

**Bug Report Archive**

**Kext Download Source**
Latest source from git, using release kext no debug.

williambj1, 2020-12-27:
Invalid report

closed 2020-12-27 by williambj1

Xodiced, 2020-12-27:
what exactly is invalid so I can correct what I did wrong?

williambj1, 2020-12-27:
Missing logs, product id, and test results using the non-airport version.