OpenIntelWireless/itlwm #474: itlwm使用一段时间断网

chwxn, 2020-12-30:
正常使用一段时间会断网，需要用HeliPort关闭后重新打开才正常。
itlwm-v1.2.0-DEBUG-alpha-4133528
mac 10.14.6  intel 9560 
下面是日志。
Log:
itlwm: iwm_tx sc->qfullmsk is FULL ring->cur=36 ring->queued=225
itlwm: iwm_tx sc->qfullmsk is FULL ring->cur=8 ring->queued=225
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: iwm_bgscan
itlwm: iwm_umac_scan
itlwm: ieee80211_setup_node 3c:46:d8:78:d1:8d
itlwm: ieee80211_setup_node 76:70:fd:d3:6a:ad
itlwm: ieee80211_setup_node 6a:db:54:a4:c7:95
itlwm: ieee80211_setup_node f6:d1:08:6c:03:d1
itlwm: ieee80211_setup_node b0:df:c1:55:de:11
itlwm: disable
itlwm: iwm_stop
itlwm: iwm_stop_device
itlwm: iwm_conf_msix_hw
itlwm: iwm_prepare_card_hw
itlwm: itlwm: RUN -> INIT
itlwm: ieee80211_set_link_state nstate=2, old_state=4
itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
syscall_extension_issue failed: 2syscall_extension_issue failed: 2itlwm: enable
itlwm: taskq task_add init_task add to queue
itlwm: iwm_init_task
itlwm: iwm_init
itlwm: iwm_init_hw
itlwm: iwm_preinit
itlwm: iwm_prepare_card_hw
itlwm: iwm_start_hw
itlwm: iwm_prepare_card_hw
itlwm: iwm_apm_init
itlwm: itlwm: L1 Disabled - LTR Enabled
itlwm: iwm_init_msix_hw
[... 162 lines cut ...]
itlwm: ieee80211_recv_4way_msg3
itlwm: ieee80211_eapol_key_mic
itlwm: itlwm: received msg 3/4 of the 4-way handshake from 68:db:54:94:c7:95
itlwm: ieee80211_send_4way_msg4
itlwm: itlwm: sending msg 4/4 of the 4-way handshake to 68:db:54:94:c7:95
itlwm: ieee80211_send_eapol_key
itlwm: ieee80211_eapol_key_mic
itlwm: ieee80211_must_update_group_key
itlwm: ieee80211_recv_4way_msg3 k_len=16 rsc=00 00 00 00 00 00itlwm: ieee80211_set_key kid=1 cipher=8, error=0
itlwm: ieee80211_set_link_state nstate=4, old_state=2
itlwm: ieee80211_set_link_state LINK_STATE_IS_UP
syscall_extension_issue failed: 2syscall_extension_issue failed: 2itlwm: taskq task_add ba_task add to queue
itlwm: itlwm: sending action to 68:db:54:94:c7:95 on channel 52 mode 11n type=208
itlwm: itlwm: sending action to 68:db:54:94:c7:95 on channel 52 mode 11n type=208
itlwm: taskq task_add ba_task add to queue
itlwm: iwm_ba_task start=1 ssn=0, tid=0 scd_queue=10
itlwm: iwm_enable_txq qid=10 tid=0 agg=1
itlwm: iwm_sta_tx_agg tx agg done. err=0 status=1, mask_status=1
itlwm: taskq task_add ba_task add to queue
itlwm: itlwm: sending action to 68:db:54:94:c7:95 on channel 52 mode 11n type=208
itlwm: itlwm: sending action to 68:db:54:94:c7:95 on channel 52 mode 11n type=208
itlwm: taskq task_add ba_task add to queue
itlwm: iwm_ba_task start=1 ssn=0, tid=2 scd_queue=12
itlwm: iwm_enable_txq qid=12 tid=2 agg=1
itlwm: iwm_sta_tx_agg tx agg done. err=0 status=1, mask_status=1

1DanielSandwich, 2020-12-30:
我也遇见过，这个可以点“DHCP续租“临时解决，不过我发现不是所有的WiFi热点都这样，比如我连接校园网公共wifi就从来不会。

chwxn, 2020-12-31:
哦，好的，我试试。。

chwxn, 2021-01-01:
> 我也遇见过，这个可以点“DHCP续租“临时解决，不过我发现不是所有的WiFi热点都这样，比如我连接校园网公共wifi就从来不会。

试了，点了续租更不行了。。

1DanielSandwich, 2021-01-01:
> > 我也遇见过，这个可以点“DHCP续租“临时解决，不过我发现不是所有的WiFi热点都这样，比如我连接校园网公共wifi就从来不会。
> 
> 试了，点了续租更不行了。。

那就不是很清楚了，我这边是有效的。可能和路由器不兼容，尝试一下其他热点。多关注一下itlwm1.2.0的更新，或许下次更新就修复了
AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-e059314
MacOS 11.1  Intel AX200

zxystd, 2021-01-01:
日志不全，要体现的是完整的从断网到重连的日志，你这里只有重连的日志，一切正常什么都看不出来。

chwxn, 2021-01-02:
> 日志不全，要体现的是完整的从断网到重连的日志，你这里只有重连的日志，一切正常什么都看不出来。

你看下这个呢？

itlwm: iwm_tx sc->qfullmsk is FULL ring->cur=188 ring->queued=225
itlwm: iwm_tx sc->qfullmsk is FULL ring->cur=241 ring->queued=225
itlwm: iwm_tx sc->qfullmsk is FULL ring->cur=152 ring->queued=225
itlwm: iwm_tx sc->qfullmsk is FULL ring->cur=198 ring->queued=225
itlwm: iwm_tx sc->qfullmsk is FULL ring->cur=200 ring->queued=225
failed to lookup target storage class: 2failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: ieee80211_free_node
itlwm: iwm_bgscan
itlwm: iwm_umac_scan
itlwm: ieee80211_setup_node 6a:db:54:a4:c7:95
itlwm: ieee80211_setup_node 74:25:8a:8b:34:90
itlwm: ieee80211_setup_node 74:25:8a:8b:32:b0
itlwm: ieee80211_setup_node f6:d1:08:6c:03:d1
itlwm: ieee80211_setup_node 3c:46:d8:78:d1:8d
itlwm: ieee80211_setup_node 36:f6:4b:c5:80:78
itlwm: ieee80211_setup_node 76:70:fd:d3:90:5a
itlwm: ieee80211_setup_node 76:70:fd:d2:ab:c2
itlwm: ieee80211_setup_node d8:c8:e9:58:42:b8
itlwm: ieee80211_setup_node b0:df:c1:55:de:11
itlwm: ieee80211_setup_node 70:f9:6d:34:be:90
itlwm: ieee80211_setup_node 2a:56:5a:49:02:22
itlwm: ieee80211_setup_node c8:e7:d8:2a:7e:d4
itlwm: ieee80211_setup_node 60:0b:03:6d:4d:50
itlwm: ieee80211_setup_node 78:44:fd:6a:52:c8
itlwm: ieee80211_setup_node 5c:b4:e2:09:11:7e
itlwm: ieee80211_setup_node 74:25:8a:8b:30:30
itlwm: ieee80211_setup_node 7c:1e:06:af:27:10
itlwm: ieee80211_setup_node 70:f9:6d:35:d1:d0
itlwm: ieee80211_setup_node 60:0b:03:de:60:f0
itlwm: ieee80211_setup_node 7c:1e:06:af:27:20
itlwm: ieee80211_setup_node 60:0b:03:de:60:e0
itlwm: ieee80211_setup_node 7c:1e:06:af:27:00
itlwm: ieee80211_setup_node 60:0b:03:6d:47:c0
itlwm: ieee80211_setup_node 60:0b:03:6d:4d:40
failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2itlwm: ieee80211_setup_node a2:af:bd:71:34:eb
itlwm: ieee80211_setup_node 70:f9:6d:34:be:80
itlwm: ieee80211_setup_node d2:ab:d5:a8:c0:a0
itlwm: ieee80211_setup_node 58:6a:b1:36:9f:60
itlwm: ieee80211_setup_node 74:25:8a:8b:2e:40
itlwm: ieee80211_setup_node d2:ab:d5:a7:f6:8e
itlwm: ieee80211_setup_node 8e:c6:81:46:41:ff
itlwm: ieee80211_setup_node 74:25:8a:8b:34:80
itlwm: ieee80211_setup_node d2:ab:d5:c9:61:5c
itlwm: ieee80211_setup_node 5e:5f:67:58:19:57
itlwm: ieee80211_setup_node 70:f9:6d:35:d1:c0
itlwm: ieee80211_setup_node 74:25:8a:85:36:a0
itlwm: ieee80211_setup_node 70:f9:6d:34:c3:60
itlwm: ieee80211_setup_node 74:25:8a:8b:32:a0
itlwm: ieee80211_setup_node 36:f6:4b:c5:58:e6
itlwm: ieee80211_setup_node d2:ab:d5:a9:62:3a
itlwm: ieee80211_setup_node 74:25:8a:8b:30:20
itlwm: ieee80211_setup_node 70:f9:6d:34:cb:40
itlwm: ieee80211_setup_node 00:0f:e2:b6:89:e0
itlwm: itlwm: end background scan
itlwm: ieee80211_node_choose_bss
itlwm: ieee80211_node_choose_bss 5ghz ssid=WiFi5 mac=68:db:54:94:c7:95 rssi=67
itlwm: ieee80211_reset_scan

chwxn, 2021-01-08:
版本：itlwm-v1.2.0-DEBUG-alpha-3f244c8.zip
目前使用这个最新的版本还是断的问题，下边的日志你看下有没有帮助。

itlwm: itlwm: end background scan
itlwm: ieee80211_node_choose_bss
itlwm: ieee80211_node_choose_bss 5ghz ssid=WiFi5 mac=68:db:54:94:c7:95 rssi=67
itlwm: ieee80211_reset_scan
Sandbox: 3 duplicate reports for deleted deny(1) file-read-data /Library/MessageTracer/SubmitDiagInfo.default.domains.searchtreeSandbox: QuickLookUIHelpe(10960) deny(1) file-read-data /Users/laptop/Library/Preferences/com.apple.LaunchServices/com.apple.launchservices.secure.plistfailed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2itlwm: iwm_bgscan
itlwm: iwm_umac_scan
itlwm: ieee80211_setup_node 64:09:80:14:c5:66
itlwm: ieee80211_setup_node a2:af:bd:43:6d:9f
itlwm: ieee80211_setup_node b0:df:c1:55:de:11
itlwm: ieee80211_setup_node 2a:56:5a:49:02:22
itlwm: ieee80211_setup_node c8:e7:d8:2a:7e:d4

.......

itlwm: itlwm: end background scan
itlwm: ieee80211_node_choose_bss
itlwm: ieee80211_node_choose_bss 5ghz ssid=WiFi5 mac=68:db:54:94:c7:95 rssi=67
itlwm: ieee80211_reset_scan
Sandbox: 3 duplicate reports for deleted deny(1) file-read-data /Library/MessageTracer/SubmitDiagInfo.default.domains.searchtreeSandbox: QuickLookUIHelpe(10960) deny(1) file-read-data /Users/laptop/Library/Preferences/com.apple.LaunchServices/com.apple.launchservices.secure.plistfailed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2syscall_extension_consume failed: 22failed to lookup target storage class: 2syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22syscall_extension_consume failed: 22failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2itlwm: iwm_bgscan
itlwm: iwm_umac_scan
itlwm: ieee80211_setup_node 64:09:80:14:c5:66
itlwm: ieee80211_setup_node a2:af:bd:43:6d:9f
itlwm: ieee80211_setup_node b0:df:c1:55:de:11
itlwm: ieee80211_setup_node 2a:56:5a:49:02:22
itlwm: ieee80211_setup_node c8:e7:d8:2a:7e:d4

> 日志不全，要体现的是完整的从断网到重连的日志，你这里只有重连的日志，一切正常什么都看不出来。

closed 2021-01-09 by chwxn