OpenIntelWireless/itlwm #477: The wifi is often disconnected, and devices connected to the same wifi lost connections at the same time

Creedowl, 2021-01-01:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: itlwm: 1.2.0 beta fw: 34.0.1
- WiFi Card Model: Intel AC 8265
- Product ID: 0x24fd
- macOS Version: 11.0.1

**Description**

After installing the recent alpha version, I encountered this issue. The wifi connection dropped constantly, and all devices connected to the same wifi lost connections for seconds. This issue seems like to be related to #419
However, the issue seems only happened on my home wifi. When I connected to an open wifi, it works well

**Bug Report Archive**

The files are exported right after disconnection 

[AirItlDebug_2021-01-01_22-07-27.zip](https://github.com/OpenIntelWireless/itlwm/files/5759030/AirItlDebug_2021-01-01_22-07-27.zip)
[Log_2021-01-01_22-12-34.log](https://github.com/OpenIntelWireless/itlwm/files/5759031/Log_2021-01-01_22-12-34.log)

**Kext Download Source**

[kext](https://github.com/OpenIntelWireless/itlwm/releases/download/v1.2.0-alpha/AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-e059314.zip)

zxystd, 2021-01-02:
have you check the MTU setting size? should be equal or lower than the ISP setting.
maybe you can get it in router?
by the way, the log is incomplete, itlwm log should be contain inside. 
[image]

Creedowl, 2021-01-02:
@zxystd I've set my MTU to 1492 (the same as what I got in router), and will check if it solves this issue. But this issue didn't occur before, just after updating to the newest airportItlwm recently.
BTW, I followed the [manual](https://openintelwireless.github.io/itlwm/Troubleshooting.html) to generate the log by running the following commands

```bash
bash <(curl -sL https://git.io/AirportItlDebug)

sudo dmesg | grep -i "itlwm" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log

```

 Is it incomplete? What show I do to get full log?

Creedowl, 2021-01-02:
The issue happened again
[AirItlDebug_2021-01-02_16-13-48.zip](https://github.com/OpenIntelWireless/itlwm/files/5759926/AirItlDebug_2021-01-02_16-13-48.zip)
[Log_2021-01-02_16-13-27.log](https://github.com/OpenIntelWireless/itlwm/files/5759927/Log_2021-01-02_16-13-27.log)

savvamitrofanov, 2021-01-04:
Same problem, I guess something wrong with taskq

paranoiagu, 2021-01-07:
I have the same issue.
here is the logs.
[logs.zip](https://github.com/OpenIntelWireless/itlwm/files/5779628/logs.zip)

StefanChimu, 2021-01-08:
Same issue for me. Big Sur 11.1 with AC-8265 and latest itlwm-1.2.0 alpha.

KrimskiyMaxim, 2021-01-11:
Same problem. 8265

rhydianbevan, 2021-01-11:
I've also got the same issue

VGerris, 2021-01-16:
I have the same issue with 1.2.0 on an AX201, it seems stable on 1.1.0.

win1010525, 2021-02-26:
Does this still happen on the latest 1.3-alpha? I think it have been fixed.

ildarnm, 2021-03-04:
Yes it happens on the latest 1.3-alpha

williambj1, 2021-03-13:
@ildarnm Please attach logs generated by the latest alpha. Also it would be nice if you can provide logs from your router as well

ildarnm, 2021-03-15:
Disconnects happen when I connected to a router at work. But it doesn't happen when I connected to a router at home.
[Report_2021-03-15_09-43-13.log](https://github.com/OpenIntelWireless/itlwm/files/6139282/Report_2021-03-15_09-43-13.log)

ronelio, 2021-03-23:
This happen often. Humax and AC12G Router. Same problem here.

The 1.3-alpha version, I noticed that is little better but still it happens.

ildarnm, 2021-04-09:
Additional info it happens each hour at 34 minutes

williambj1, 2021-10-02:
Hi all, does the same behavior still occur when using the latest release?

otalonym, 2021-11-16:
544.348151]: itlwm: setAWDL_SYNCHRONIZATION_CHANNEL_SEQUENCE
[  544.348155]: itlwm: dumpAWDLChannelSeqs length

ildarnm, 2021-11-23:
I don't see this behavior anymore.

VGerris, 2021-11-23:
Same here, seems ok now

williambj1, 2022-01-18:
Thanks for the feedback.

closed 2022-01-18 by williambj1