OpenIntelWireless/itlwm #479: 可以搜索到网络但是一直提示输入密码

jiaxin96, 2021-01-02:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.2-alpha
- WiFi Card Model: Ax200
- Product ID: 华硕B460i
- macOS Version: 
11.1
**Description**
可以搜索到网络但是一直提示输入密码

**Bug Report Archive**
[Report_2021-01-03_00-02-42.log](https://github.com/OpenIntelWireless/itlwm/files/5760510/Report_2021-01-03_00-02-42.log)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/commit/e059314af91cae4ce514d69cf900c64b7a78f4e6

wxyttty, 2021-01-02:
这是一个很玄学的问题，有人是改驱动加载顺序解决的，我是去掉-v解决的，然后再加上-v也还可以连接了

zxystd, 2021-01-04:
密码错误？

wqianglxf, 2021-01-04:
我的也是这样，但不连接5G WiFi的话可以使用。连接5g时，有数据传输时会中断提示再连或要求输入密码。网卡是7265

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: zxystd <notifications@github.com&gt;
发送时间: 2021年1月4日 08:45
收件人: OpenIntelWireless/itlwm <itlwm@noreply.github.com&gt;
抄送: Subscribed <subscribed@noreply.github.com&gt;
主题: 回复：[OpenIntelWireless/itlwm] 可以搜索到网络但是一直提示输入密码 (#479)

wxyttty, 2021-01-04:
这不是密码错误的问题，我的ax200不管是2.4还是5g都不行，连加密类型都会判断错误，aes的显示是tkis，无加密的也会提示输入密码，只能连上默认的wifi，并且断开后无法再连回。

jiaxin96, 2021-01-04:
> 这不是密码错误的问题，我的ax200不管是2.4还是5g都不行，连加密类型都会判断错误，aes的显示是tkis，无加密的也会提示输入密码，只能连上默认的wifi，并且断开后无法再连回。

我有幸调整加载顺序成功连接一次，不过重启后就又不行了

jiaxin96, 2021-01-04:
> 密码错误？

不是密码的问题，也不是路由器的问题，我尝试过很多路由器了，2.4和5G都有，开发的网络也无法连接。而且这个需要怎么重置呢，我在系统wifi设置忘记网络，他好像不会改变首选项。

jiaxin96, 2021-01-04:
> **Have You Read Our Docs**
> Yes
> 
> **Are You Reporting A Bug**
> Yes
> 
> **Environment**
> 
> * Kext Version: 1.2-alpha
> * WiFi Card Model: Ax200
> * Product ID: 华硕B460i
> * macOS Version:
>   11.1
>   **Description**
>   可以搜索到网络但是一直提示输入密码
> 
> **Bug Report Archive**
> [Report_2021-01-03_00-02-42.log](https://github.com/OpenIntelWireless/itlwm/files/5760510/Report_2021-01-03_00-02-42.log)
> 
> **Kext Download Source**
> [e059314](https://github.com/OpenIntelWireless/itlwm/commit/e059314af91cae4ce514d69cf900c64b7a78f4e6)
补充说明一下，我的OC版本是0.65

juniorzhou, 2021-01-05:
同系统版本。同AX200。 kext，3f244c8。 
同样的问题，一直提示输入密码。

jiaxin96, 2021-01-06:
https://github.com/a565109863/AirPortOpenBSD/releases/tag/v2.0.6
可以先用这个代替，目前11.1下稳定使用

zxystd, 2021-01-06:
从哪个版本开始有这个问题？11.1？AiportItlwm.kext使用的是IO80211Family的原生认证功能（未公开的API），可能随着系统升级之后进行了些API变更，有时间我更新一个版本试一下。也请提供完整的日志，日志收集方法在文档里有。

jiaxin96, 2021-01-06:
> 从哪个版本开始有这个问题？11.1？AiportItlwm.kext使用的是IO80211Family的原生认证功能（未公开的API），可能随着系统升级之后进行了些API变更，有时间我更新一个版本试一下。也请提供完整的日志，日志收集方法在文档里有。

OK，按照我收集的信息是11.1这个版本出问题的人最多，我现在使用的是https://github.com/a565109863/AirPortOpenBSD/releases/tag/v2.0.6，目前正常使用，您的驱动日志信息等我明天提供一份详细的在这个issue下。

jiaxin96, 2021-01-06:
[    0.587660]: AirportItlwm::init AirPortFamily_kexts-1838.2 "AirPortFamily_kexts-1838.2" Dec  2 2020 20:14:59
[    1.045419]: itlwm: iwx_prepare_card_hw
[    1.045423]: itlwm: iwx_set_hw_ready
[    1.099317]: itlwm: taskq taskq_create_thread lock
[    1.099319]: itlwm: taskq taskq_create_thread lock
[    1.099321]: itlwm: taskq taskq_create_thread unlock
[    1.099321]: itlwm: taskq taskq_create_thread unlock
[    1.099324]: itlwm: ieee80211_channel_init
[    1.099325]: itlwm: taskq taskq_create_thread lock
[    1.099325]: itlwm: taskq taskq_create_thread lock
[    1.099326]: itlwm: taskq taskq_create_thread unlock
[    1.099327]: itlwm: taskq taskq_create_thread unlock
[    1.099328]: itlwm: taskq taskq_thread schedule task
[    1.099329]: itlwm: ieee80211_set_link_state nstate=2, old_state=-1
[    1.099330]: itlwm: taskq taskq_thread schedule task
[    1.099331]: itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
[    1.099333]: itlwm: ieee80211_media_init
[    1.099339]: itlwm: iwx_prepare_card_hw
[    1.099341]: itlwm: iwx_set_hw_ready
[    1.099357]: itlwm: iwx_start_hw
[    1.099358]: itlwm: iwx_prepare_card_hw
[    1.099359]: itlwm: iwx_set_hw_ready
[    1.110982]: itlwm: : L1 Enabled - LTR Enabled
[    1.111182]: itlwm: iwx_set_hw_ready
[    1.111196]: itlwm: iwx_run_init_mvm_ucode
[    1.111198]: itlwm: iwx_load_ucode_wait_alive
[    1.115786]: itlwm: load firmware iwlwifi-cc-a0-48.ucode done
[    1.115789]: itlwm: parsing firmware iwlwifi-cc-a0-48.ucode
[    1.115794]: itlwm: : ucode type 0 section 0
[    1.115796]: itlwm: : ucode type 0 section 1
[    1.115797]: itlwm: : ucode type 0 section 2
[    1.115799]: itlwm: : ucode type 0 section 3
[    1.115800]: itlwm: : ucode type 0 section 4
[    1.115802]: itlwm: : ucode type 0 section 5
[    1.115803]: itlwm: : ucode type 0 section 6
[    1.115805]: itlwm: : ucode type 0 section 7
[    1.115806]: itlwm: : ucode type 0 section 8
[    1.115808]: itlwm: : ucode type 0 section 9
[    1.115809]: itlwm: : ucode type 0 section 10
[    1.115811]: itlwm: : ucode type 0 section 11
[    1.115812]: itlwm: : ucode type 0 section 12
[    1.115814]: itlwm: : ucode type 0 section 13
[    1.115816]: itlwm: : ucode type 0 section 14
[    1.115818]: itlwm: : ucode type 0 section 15
[    1.115819]: itlwm: : ucode type 0 section 16
[    1.115821]: itlwm: : ucode type 0 section 17
[    1.115822]: itlwm: : ucode type 0 section 18
[    1.115824]: itlwm: : ucode type 0 section 19
[    1.115825]: itlwm: : ucode type 0 section 20
[    1.115827]: itlwm: : ucode type 0 section 21
[    1.115828]: itlwm: : ucode type 0 section 22
[    1.115830]: itlwm: : ucode type 0 section 23
[    1.115831]: itlwm: : ucode type 0 section 24
[    1.115833]: itlwm: : ucode type 0 section 25
[    1.115834]: itlwm: : ucode type 0 section 26
[    1.115836]: itlwm: : ucode type 0 section 27
[    1.115837]: itlwm: : ucode type 0 section 28
[    1.115839]: itlwm: : ucode type 0 section 29
[    1.115840]: itlwm: : ucode type 0 section 30
[    1.115842]: itlwm: : ucode type 0 section 31
[    1.115843]: itlwm: : ucode type 0 section 32
[    1.115845]: itlwm: : ucode type 0 section 33
[    1.115846]: itlwm: : ucode type 0 section 34
[    1.115848]: itlwm: : ucode type 0 section 35
[    1.115849]: itlwm: : ucode type 0 section 36
[    1.115851]: itlwm: : ucode type 0 section 37
[    1.115852]: itlwm: : ucode type 0 section 38
[    1.115854]: itlwm: : ucode type 0 section 39
[    1.115855]: itlwm: : ucode type 0 section 40
[    1.115857]: itlwm: : ucode type 0 section 41
[... 441 lines cut ...]
[   39.069879]: itlwm: setCIPHER_KEYitlwm: Set key request: len=0 cipher_type=0 flags=0 index=3 key= rsc_len=0 rsc= ea=00 00 00 00 00 00
[   39.294930]: itlwm: : authentication timed out for c8:e7:d8:d4:b7:2a
[   39.299705]: itlwm: iwx_add_task iwx_newstate_task
[   39.303070]: itlwm: taskq task_add iwx_newstate_task add to queue
[   39.307732]: itlwm: ieee80211_watchdog 903 nstate=1
[   39.307764]: itlwm: iwx_newstate_task
[   39.307767]: itlwm: iwx_newstate_task sc->sc_flags & IWX_FLAG_SHUTDOWN false
[   39.307769]: itlwm: iwx_deauth
[   39.307771]: itlwm: iwx_send_cmd: sending command 0x1e
[   39.308007]: itlwm: iwx_cmd_done: command 0x1e done
[   39.308026]: itlwm: iwx_send_cmd: sending command 0x19
[   39.308923]: itlwm: iwx_cmd_done: command 0x19 done
[   39.308929]: itlwm: iwx_send_cmd_pdu_status
[   39.308932]: itlwm: iwx_send_cmd: sending command 0x2b
[   39.309086]: itlwm: iwx_cmd_done: command 0x2b done
[   39.309097]: itlwm: iwx_send_cmd: sending command 0x28
[   39.309245]: itlwm: iwx_cmd_done: command 0x28 done
[   39.309250]: itlwm: iwx_scan
[   39.309251]: itlwm: iwx_umac_scan
[   39.309257]: itlwm: iwx_send_cmd: sending command 0x10d
[   39.309800]: itlwm: iwx_cmd_done: command 0x10d done
[   39.309806]: itlwm: ieee80211_reset_scan
[   39.309808]: itlwm: : AUTH -> SCAN
[   39.309809]: itlwm: ieee80211_set_link_state nstate=2, old_state=2
[   39.976951]: itlwm: ieee80211_setup_node 06:74:9c:6e:9a:0a

sungoku, 2021-01-14:
> 这是一个很玄学的问题，有人是改驱动加载顺序解决的，我是去掉-v解决的，然后再加上-v也还可以连接了

请问你是怎么修改的加载顺序？我也是AX200的卡，目前是5G能连，2.4G连不上，反复要求输入密码。 @wxyttty

jiaxin96, 2021-01-14:
直接用最新版正式版，我现在可以用了。

closed 2021-01-14 by jiaxin96

delete-user, 2021-09-25:
不知道是不是同一个问题，发现只要密码里带字符就反复要求输入密码