OpenIntelWireless/itlwm #480: 系统睡眠唤醒时panic

Just-Prog, 2021-01-04:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: Github-Action最新版本
- WiFi Card Model: AC3165
- Product ID: MacBookPro15,2
- macOS Version: 1.15.7

**Description**

~~~
panic(cpu 5 caller 0xffffff7fa27f9e97): itlwm: mac clock not ready
~~~

是EFI的问题还是itlwm自身的问题呢？

**Kext Download Source**

Github Release

williambj1, 2021-01-04:
Please provide complete logs

Just-Prog, 2021-01-04:
> Please provide complete logs

~~~
panic(cpu 5 caller 0xffffff7f943f9e97): itlwm: mac clock not ready
Backtrace (CPU 5), Frame : Return Address
0xffffff819543ba10 : 0xffffff801131a65d 
0xffffff819543ba60 : 0xffffff8011454a75 
0xffffff819543baa0 : 0xffffff80114465fe 
0xffffff819543baf0 : 0xffffff80112c0a40 
0xffffff819543bb10 : 0xffffff8011319d27 
0xffffff819543bc10 : 0xffffff801131a117 
0xffffff819543bc60 : 0xffffff8011ac1a6c 
0xffffff819543bcd0 : 0xffffff7f943f9e97 
0xffffff819543bd00 : 0xffffff7f943fa146 
0xffffff819543bd40 : 0xffffff7f943ff701 
0xffffff819543bd90 : 0xffffff7f9441f80e 
0xffffff819543bdf0 : 0xffffff7f94424177 
0xffffff819543be40 : 0xffffff7f9438885f 
0xffffff819543bfa0 : 0xffffff80112c013e 
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(1.2)[ED2865EE-FEA7-309D-BCC0-00952539C9C3]@0xffffff7f94388000->0xffffff7f95065fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[44472E6F-8DA0-3B46-ADEF-AFF76EC6C6DB]@0xffffff7f920b4000
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[D6EB9626-5A80-3FE4-A1F3-4F455A48C64A]@0xffffff7f9421e000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[26FE14A5-825D-35E4-BD06-C5B8A1AE1FD9]@0xffffff7f91d6f000

BSD process name corresponding to current thread: kernel_task
Boot args: -lilubetaall -igfxdump debug=0x100 -disablegfxfirmware -v vsmcbeta 

Mac OS version:
19H2

Kernel version:
Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64
Kernel UUID: 05D51A3D-3A87-3FF0-98C3-9CF3827A3EDD
Kernel slide:     0x0000000011000000
Kernel text base: 0xffffff8011200000
__HIB  text base: 0xffffff8011100000
System model name: MacBookPro15,2 (Mac-827FB448E656EC26)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 2454727681334
last loaded kext at 18085973647: >AudioAUUC	1.70 (addr 0xffffff7f93b3f000, size 32768)
last unloaded kext at 424992596467: >!A!ILpssGspi	3.0.60 (addr 0xffffff7f932db000, size 45056)
loaded kexts:
com.paragon-software.filesystems.ntfs	143.5.15
com.zxystd.!I!BFirmware	1.1.2
me.kishorprins.VoodooInput	1.1.0
as.acidanthera.voodoo.driver.PS2Keyboard	2.1.9
com.zxystd.AirportItlwm	1.2.0
as.acidanthera.voodoo.driver.PS2!C	2.1.9
org.rehabman.driver.NullEthernet	1.0.6
com.alexandred.VoodooI2CHID	1
com.alexandred.VoodooI2C	2.6.2
org.coolstar.VoodooGPIO	1.1
as.vit9696.SMCProcessor	1.1.9
com.rehabman.driver.USBInjectAll	0.7.6
ru.usrsse2.SMCBatteryManager	1.1.9
as.vit9696.VirtualSMC	1.1.9
org.acidanthera.NVMeFix	1.0.4
xyz.racermaster.NoTouchID	1.0.4
as.lvs1974.HibernationFixup	1.3.8
org.vanilla.driver.CPUFriend	1.2.2
as.acidanthera.BrightnessKeys	1.0.1
com.alexandred.VoodooI2CServices	1
as.vit9696.!AALC	1.5.5
as.vit9696.WhateverGreen	1.4.5
as.vit9696.Lilu	1.5.0
>AudioAUUC	1.70
@fileutil	20.036.15
@filesystems.autofs	3.0
>!ASmartBatteryManager	161.0.0
>!AUpstreamUserClient	3.6.8
>!AHDA	283.15
>!AGraphicsDevicePolicy	5.2.6
@AGDCPluginDisplayMetrics	5.2.6
@private.KextAudit	1.0
>!AHV	1
|IOUserEthernet	1.0.1
|IO!BSerialManager	7.0.6f7
>pmtelemetry	1
@Dont_Steal_Mac_OS_X	7.0.0
>!A!IKBLGraphics	14.0.7
>ACPI_SMC_PlatformPlugin	1.0.0
>!A!ISlowAdaptiveClocking	4.0.0
>eficheck	1
>!A!IPCHPMC	2.0.1
>!A!IKBLGraphicsFramebuffer	14.0.7
>!ABacklight	180.3
>!AMCCSControl	1.14
>!ATopCaseHIDEventDriver	3430.1
>!AVirtIO	1.0
@filesystems.hfs.kext	522.100.5
@!AFSCompression.!AFSCompressionTypeDataless	1.0.0d1
@BootCache	40
@!AFSCompression.!AFSCompressionTypeZlib	1.0.0
@filesystems.apfs	1412.141.1
>!AAHCIPort	341.140.1
>!AHPET	1.8
>!ARTC	2.0
>!AACPIButtons	6.1
>!AACPIEC	6.1
>!ASMBIOS	2.1
>!AAPIC	1.7
$!AImage4	1
@nke.applicationfirewall	303
$TMSafetyNet	8
@!ASystemPolicy	2.0.0
|EndpointSecurity	1
@kext.triggers	1.0
>DspFuncLib	283.15
@kext.OSvKernDSPLib	529
>!AGraphicsControl	5.2.6
|IOAVB!F	850.1
>!AHDA!C	283.15
|IOHDA!F	283.15
@!AGPUWrangler	5.2.6
>IOPlatformPluginLegacy	1.0.0
|IOSlowAdaptiveClocking!F	1.0.0
>IOPlatformPlugin!F	6.0.0d8
@!AGraphicsDeviceControl	5.2.6
|IOAccelerator!F2	438.7.3
>!ABacklightExpert	1.1.0
|IONDRVSupport	576.1
>!ASMBus!C	1.0.18d1
|IOGraphics!F	576.1
@plugin.IOgPTPPlugin	840.3
|IOEthernetAVB!C	1.1.0
>usb.networking	5.0.0
>usb.!UHostCompositeDevice	1.2
|Broadcom!BHost!CUSBTransport	7.0.6f7
|IO!BHost!CUSBTransport	7.0.6f7
|IO!BHost!CTransport	7.0.6f7
>!AActuatorDriver	3440.1
>!AHS!BDriver	3430.1
>IO!BHIDDriver	7.0.6f7
|IO!B!F	7.0.6f7
|IO!BPacketLogger	7.0.6f7
>!AMultitouchDriver	3440.1
>!AInputDeviceSupport	3440.8
|IOAudio!F	300.2
@vecLib.kext	1.2.0
|IOSerial!F	11
|IOSurface	269.11
@filesystems.hfs.encodings.kext	1
>usb.!UHostPacketFilter	1.0
|IOUSB!F	900.4.2
|IOAHCIBlock!S	316.100.5
@hack.HackrNVMe!F	92.1.0
|IO80211!F	1200.12.2b1
|IOSkywalk!F	1
>corecapture	1.0.4
|IOAHCI!F	290.0.1
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
|IOSMBus!F	1.1
|IOHID!F	2.0.0
>DiskImages	493.0.0
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!ASSE	1.0
>!AKeyStore	2
>!UTDM	489.120.1
|IOSCSIBlockCommandsDevice	422.120.3
>!ACredentialManager	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.140.1
|IOSCSIArchitectureModel!F	422.120.3
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	840.3
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

~~~

williambj1, 2021-01-04:
Please add `keepsyms=1` to `boot-args` and report again.

Just-Prog, 2021-01-04:
> Please add `keepsyms=1` to `boot-args` and report again.

新LOG抓到了
辛苦您了。

~~~
panic(cpu 0 caller 0xffffff7f9e1f9e97): itlwm: mac clock not ready
Backtrace (CPU 0), Frame : Return Address
0xffffff81a7793a10 : 0xffffff801b11a65d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff81a7793a60 : 0xffffff801b254a75 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81a7793aa0 : 0xffffff801b2465fe mach_kernel : _kernel_trap + 0x4ee
0xffffff81a7793af0 : 0xffffff801b0c0a40 mach_kernel : _return_from_trap + 0xe0
0xffffff81a7793b10 : 0xffffff801b119d27 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff81a7793c10 : 0xffffff801b11a117 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff81a7793c60 : 0xffffff801b8c1a6c mach_kernel : _panic + 0x54
0xffffff81a7793cd0 : 0xffffff7f9e1f9e97 com.zxystd.AirportItlwm : __ZN6ItlIwm21iwm_nic_assert_lockedEP9iwm_softc + 0x57
0xffffff81a7793d00 : 0xffffff7f9e1fa146 com.zxystd.AirportItlwm : __ZN6ItlIwm12iwm_nic_lockEP9iwm_softc + 0x36
0xffffff81a7793d40 : 0xffffff7f9e1ff701 com.zxystd.AirportItlwm : __ZN6ItlIwm15iwm_stop_deviceEP9iwm_softc + 0x61
0xffffff81a7793d90 : 0xffffff7f9e21f80e com.zxystd.AirportItlwm : __ZN6ItlIwm8iwm_stopEP6_ifnet + 0x10e
0xffffff81a7793df0 : 0xffffff7f9e224177 com.zxystd.AirportItlwm : __ZN6ItlIwm13iwm_init_taskEPv + 0xc7
0xffffff81a7793e40 : 0xffffff7f9e18885f com.zxystd.AirportItlwm : __Z12taskq_threadPv + 0x6f
0xffffff81a7793fa0 : 0xffffff801b0c013e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(1.2)[ED2865EE-FEA7-309D-BCC0-00952539C9C3]@0xffffff7f9e188000->0xffffff7f9eec0fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[44472E6F-8DA0-3B46-ADEF-AFF76EC6C6DB]@0xffffff7f9beb4000
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[D6EB9626-5A80-3FE4-A1F3-4F455A48C64A]@0xffffff7f9e01e000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[26FE14A5-825D-35E4-BD06-C5B8A1AE1FD9]@0xffffff7f9bb6f000

BSD process name corresponding to current thread: kernel_task
Boot args: -lilubetaall -igfxdump debug=0x100 -disablegfxfirmware -v vsmcbeta keepsyms=1 

Mac OS version:
19H2

Kernel version:
Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64
Kernel UUID: 05D51A3D-3A87-3FF0-98C3-9CF3827A3EDD
Kernel slide:     0x000000001ae00000
Kernel text base: 0xffffff801b000000
__HIB  text base: 0xffffff801af00000
System model name: MacBookPro15,2 (Mac-827FB448E656EC26)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 138156582015
last loaded kext at 46581142329: >!UAudio	323.4 (addr 0xffffff7fa03a6000, size 434176)
loaded kexts:
com.paragon-software.filesystems.ntfs	143.5.15
com.zxystd.!I!BFirmware	1.1.2
me.kishorprins.VoodooInput	1.1.0
as.acidanthera.voodoo.driver.PS2Mouse	2.1.9
as.acidanthera.voodoo.driver.PS2Trackpad	2.1.9
as.acidanthera.voodoo.driver.PS2Keyboard	2.1.9
com.zxystd.AirportItlwm	1.2.0
as.acidanthera.voodoo.driver.PS2!C	2.1.9
org.rehabman.driver.NullEthernet	1.0.6
com.alexandred.VoodooI2CHID	1
com.alexandred.VoodooI2C	2.6.2
org.coolstar.VoodooGPIO	1.1
as.vit9696.SMCProcessor	1.1.9
com.rehabman.driver.USBInjectAll	0.7.6
ru.usrsse2.SMCBatteryManager	1.1.9
as.vit9696.VirtualSMC	1.1.9
org.acidanthera.NVMeFix	1.0.4
xyz.racermaster.NoTouchID	1.0.4
as.lvs1974.HibernationFixup	1.3.8
org.vanilla.driver.CPUFriend	1.2.2
as.acidanthera.BrightnessKeys	1.0.1
com.alexandred.VoodooI2CServices	1
as.vit9696.!AALC	1.5.5
[... 117 lines cut ...]
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.140.1
|IOSCSIArchitectureModel!F	422.120.3
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	840.3
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

~~~

zxystd, 2021-01-07:
EFI的问题，设备未准备好，睡死了。

RANSAA, 2021-01-11:
> EFI的问题，设备未准备好，睡死了。

这个呢？
`
panic(cpu 4 caller 0xffffff7fa3db7fcd): itlwm: nic locks counter 0
Backtrace (CPU 4), Frame : Return Address
0xffffff914518b8d0 : 0xffffff802011961d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff914518b920 : 0xffffff8020253ac5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff914518b960 : 0xffffff802024564e mach_kernel : _kernel_trap + 0x4ee
0xffffff914518b9b0 : 0xffffff80200bfa40 mach_kernel : _return_from_trap + 0xe0
0xffffff914518b9d0 : 0xffffff8020118ce7 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff914518bad0 : 0xffffff80201190d7 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff914518bb20 : 0xffffff80208c00bc mach_kernel : _panic + 0x54
0xffffff914518bb90 : 0xffffff7fa3db7fcd com.zxystd.AirportItlwm : __ZN6ItlIwm21iwm_nic_assert_lockedEP9iwm_softc + 0xad
0xffffff914518bbc0 : 0xffffff7fa3dbb36e com.zxystd.AirportItlwm : __ZN6ItlIwm14iwm_enable_txqEP9iwm_softciiiiii + 0x8e
0xffffff914518bc70 : 0xffffff7fa3de4622 com.zxystd.AirportItlwm : __ZN6ItlIwm15iwm_add_aux_staEP9iwm_softc + 0x82
0xffffff914518bd00 : 0xffffff7fa3dde6df com.zxystd.AirportItlwm : __ZN6ItlIwm11iwm_init_hwEP9iwm_softc + 0x2ff
0xffffff914518bd70 : 0xffffff7fa3dddb46 com.zxystd.AirportItlwm : __ZN6ItlIwm8iwm_initEP6_ifnet + 0xd6
0xffffff914518bdf0 : 0xffffff7fa3de2230 com.zxystd.AirportItlwm : __ZN6ItlIwm13iwm_init_taskEPv + 0x110
0xffffff914518be40 : 0xffffff7fa3d465cf com.zxystd.AirportItlwm : __Z12taskq_threadPv + 0x6f
0xffffff914518bfa0 : 0xffffff80200bf13e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(1.2)[21FB2ADA-E805-3187-B6F4-6824206CD1E9]@0xffffff7fa3d45000->0xffffff7fa4a83fff
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[CEFF979E-2922-3260-BF47-49AEAAF78C4A]@0xffffff7fa1a6c000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[B3167785-465D-34E5-B7C3-D392C9431A26]@0xffffff7fa0eb0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[0F12857B-4B0E-3060-9519-2B65B17CC60C]@0xffffff7fa0b12000

BSD process name corresponding to current thread: kernel_task
Boot args: keepsyms=1 debug=0x100 npci=0x2000 -lilubetaall -wegnoegpu chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
19H114

Kernel version:
Darwin Kernel Version 19.6.0: Tue Nov 10 00:10:30 PST 2020; root:xnu-6153.141.10~1/RELEASE_X86_64
Kernel UUID: A477D537-F87E-3978-8D7F-A84B7C02E62E
Kernel slide:     0x000000001fe00000
Kernel text base: 0xffffff8020000000
__HIB  text base: 0xffffff801ff00000
System model name: MacBookPro13,3 (Mac-A5C67F76ED83108C)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 4161219167413

`

williambj1, 2021-01-16:
This issue is usually caused by misconfigurations that causes instant wakes when the device is not ready.

closed 2021-01-16 by williambj1

iflovewind, 2021-01-19:
解决了吗，我也遇到同样的问题

RANSAA, 2021-01-19:
> 解决了吗，我也遇到同样的问题

没有

iflovewind, 2021-01-20:
> > 解决了吗，我也遇到同样的问题
> 
> 没有

今天试了一下，在系统偏好设置、节能、去掉唤醒以供wi-fi访问貌似 不会睡死，可以去试试