OpenIntelWireless/itlwm #481: 来个打赏二维码。

han123580, 2021-01-05:
大佬请贴上打赏二维码。十分想感谢您和您的团队

silentchild, 2021-01-05:
支持！

wqianglxf, 2021-01-05:
支持

zxystd, 2021-01-16:
谢谢支持。目前我们并没有接受现金捐赠的计划。如之前所说，我们需要各种路由器、网卡、电脑上做适配兼容，虽然我们自己购买了一些设备，也建立了内部测试体验QQ群，但是我们不可能把所有类型的设备都买到并考虑到位，所以最好的办法还是通过大家的测试及按照文档的指引正确给出相应的反馈，这样才能让驱动变得更好，这也是贡献的一种方式。当然如果有大佬愿意并且有能力一起参与开发也欢迎联系我，当然直接上代码那就更好了。

win1010525, 2021-02-26:
Group testing is more efficient.

williamoak1988, 2021-03-30:
支持开通打赏功能，开发团队做了这么大贡献，感谢

chanbaihai, 2021-08-17:
+1 开通打赏吧 这个驱动省去了大部分人换网卡的钱

HeartAndSoul001, 2022-03-09:
> 谢谢支持。目前我们并没有接受现金捐赠的计划。如之前所说，我们需要各种路由器、网卡、电脑上做适配兼容，虽然我们自己购买了一些设备，也建立了内部测试体验QQ群，但是我们不可能把所有类型的设备都买到并考虑到位，所以最好的办法还是通过大家的测试及按照文档的指引正确给出相应的反馈，这样才能让驱动变得更好，这也是贡献的一种方式。当然如果有大佬愿意并且有能力一起参与开发也欢迎联系我，当然直接上代码那就更好了。

大佬，我想问一下，你是怎么入门Opencore的项目的？你这个驱动是在OC项目的架构上开发的，肯定吃透了OC的相关接口。我想请教一下需要哪些知识体系，我平时也偶尔有些空余时间，也有c++的开发基础，接触过嵌入式开发