OpenIntelWireless/itlwm #482: 7265AC蓝牙不能和WIFI一起用BUG反馈

zncs520, 2021-01-05:
我有2个机器(NUC8和GPD P2MAX)
NUC8的9560AC基本没有问题,  WIFI 和蓝牙都很完美(就是4.0的蓝牙鼠标连不上)

P2MAX的7265AC蓝牙不能和WIFI一起用
一起用的话蓝牙基本连不上外设(非常不稳定,容易断或者连不上)
关闭WIFI后, 蓝牙就正常了

wqianglxf, 2021-01-05:
7265ac，这个蓝牙我也没有搞定，和你的情况差不多

vimcaw, 2021-01-12:
修改一下加载顺序试一下：https://github.com/OpenIntelWireless/itlwm/issues/291

klzbw, 2021-01-15:
Wi-Fi与蓝牙都2.4G,目前没什么好办法解决干扰问题！

royatgh, 2021-02-20:
我是根本不显示蓝牙，wifi可用就是慢，（bluetoothinject/bluetoothFirmware/airportitlwm）顺序怎么调都是一样结果：看不到蓝牙。而且“log show --last boot | grep IntelFirmware”无日志。

Environment

Kext Version: AirportItlwm_v1.2.0_stable_BigSur.kext.zip
WiFi Card Model: Intel Dual Band Wireless AC 7265   （NUC5i7RYH原配Wi-Fi）
Product ID: 这个不知道怎么获取
macOS Version: BigSur 11.2.1

win1010525, 2021-02-20:
> 我是根本不显示蓝牙，wifi可用就是慢，（bluetoothinject/bluetoothFirmware/airportitlwm）顺序怎么调都是一样结果：看不到蓝牙。而且“log show --last boot | grep IntelFirmware”无日志。
>
> Environment
>
> Kext Version: [...]

你是当时群里问我的那个人吗

win1010525, 2021-02-20:
> 你是当时群里问我的那个人吗

如果是的话，你驱动根本没加载，自己调整efi设置
Your kext didn't load at all. Adjust your efi settings.

win1010525, 2021-02-20:
> 我是根本不显示蓝牙，wifi可用就是慢，（bluetoothinject/bluetoothFirmware/airportitlwm）顺序怎么调都是一样结果：看不到蓝牙。而且“log show --last boot | grep IntelFirmware”无日志。
>
> Environment
>
> Kext Version: [...]

Sorry but that's not you. Wi-Fi performance low is a known issue. If there is no log for bluetooth, that means your kext didn't load at all. You can confirm it with `kextstat`.

win1010525, 2021-02-20:
> 我有2个机器(NUC8和GPD P2MAX)
> NUC8的9560AC基本没有问题, WIFI 和蓝牙都很完美(就是4.0的蓝牙鼠标连不上)
> 
> P2MAX的7265AC蓝牙不能和WIFI一起用
> 一起用的话蓝牙基本连不上外设(非常不稳定,容易断或者连不上)
> 关闭WIFI后, 蓝牙就正常了

连不上外设是驱动的锅，已知了

win1010525, 2021-02-20:
你可以买苹果的外设，能连上（🐶保命）

xjy37-me, 2021-04-03:
The same boat. Bluethooth device can’t connect well on my 7265AC, ASUS Laptop, while it works fine when Wifi is off. I’ve tried different orders of Airportitwm & IntelBluethooth kexts but no sense.

zxystd, 2021-05-07:
hmmm, I don't have this card... ac 7265/7260 have lots of variants, some works fine while others not, I don't know if I buy one of them can reproduce the issue.

sirius-2, 2021-06-08:
同样的7265AC，买了免驱USB蓝牙在macOS BigSur上同样出现了这样的问题。即USB蓝牙能正常开关，但是连上后几秒断开。不确定是否是Airportitlwm自身的问题，以我的体验原生板载蓝牙适配更好至少能连上虽然不稳定听歌。供参考 @zxystd

wqianglxf, 2021-06-08:
我现在用的是绿联USB蓝牙对应方案。可以正常用鼠标、耳机、键盘（罗技375每次要手动连）。内置的蓝牙和WIFI不能同时用，不然蓝牙连不上或者连上就掉线了。

zxystd, 2021-07-21:
[ac7265.zip](https://github.com/OpenIntelWireless/itlwm/files/6853576/ac7265.zip)
Is there anyone have BT/WIFI coex problems with 7265 can help me to test? using the kext and see the firmware version.

sirius-2, 2021-07-22:
笔记本可以吗，刚好有这个型号

sirius-2, 2021-07-22:
Bluetooth Firmware Version : `8.4096`

AirportItlwm: `AirportItlwm_7265.kext`

Bluetooth Driver updated to the newest as 1.1.3.

[image]

Status: Disconnection still appears after pair successfully.
[7265.log](https://github.com/OpenIntelWireless/itlwm/files/6860105/7265.log)

zxystd, 2021-07-22:
@sirius-2 Thanks, can you also capture the wifi info?

sirius-2, 2021-07-22:
Absolutely, please **feel free** to ask for more details for debugging.
[image]

> No more useful info. for wifi list following.

zxystd, 2021-07-22:
So your card variant is 7265D, before, we don't distinguish between 7265 and 7265D and they are using the same firmware, but yesterday I am doubting if the coex issue is related to the different firmware(I think there is some hardware bugs in 7265 variants, so Intel make different firmwares to them), so I change to code like before:
[image]
But according to your description it behave the same, still buggy.

sirius-2, 2021-07-22:
The key point is to find out the right firmware for each of 7265 variants. Isn't it ? I don't have enough knowledge about how it works yet. So you just try it. 

Remind me if any problems or breakthrough process. Thanks for your work again !

zxystd, 2021-07-22:
@sirius-2 Linux iwlwifi distinguish the firmwares, I don't know why OpenBSD code not doing this(but the newest code is changed), so I just have a try.
Another question, when did the Bluetooth not work? when the wifi is scanning? or connected?

sirius-2, 2021-07-22:
@zxystd Whenever Wifi is turned off or not, bluetooth disconnects after several seconds as it sends pair request and paired successfully. Result just takes responsibility for test  of this time.
> [For reference](https://github.com/sirius-2/OpenCore-EFI-ASUS)

zxystd, 2021-07-23:
[ac7265_enable_coex_flags.zip](https://github.com/OpenIntelWireless/itlwm/files/6867779/ac7265_enable_coex_flags.zip)
@sirius-2 How about this one?

sirius-2, 2021-07-23:
> nvram reset done

Bluetooth Firmware Version : `8.4096`

[image]

[image]

> Status: It seems last more time before disconnect that is  from 2-3 seconds to 4 seconds
[7265_1.log](https://github.com/OpenIntelWireless/itlwm/files/6868899/7265_1.log)

sirius-2, 2021-07-23:
@zxystd Feel free to learn more details. 
> Status: It disconnects whenever wifi is turned on or not.

zxystd, 2021-07-23:
[ac7265D_2.zip](https://github.com/OpenIntelWireless/itlwm/files/6869288/ac7265D_2.zip)
@sirius-2 please try this one.

sirius-2, 2021-07-23:
@zxytsd Well, there is no change but log changes.
> Status: It disconnects after 4 seconds as connected.

[ac7265D_2.log](https://github.com/OpenIntelWireless/itlwm/files/6869726/ac7265D_2.log)

wqianglxf, 2021-07-24:
我有台电脑和@zncs520 一样GPD P2 Max，7265AC的网卡。我用了最新的蓝牙和itlwm驱动，wifi可以正常使用，Airportitlwm无法热点。蓝牙可以搜索到设备，但无法正常连接上。连接后几秒后就会断开。
[image]
[image]
[ac7265.log](https://github.com/OpenIntelWireless/itlwm/files/6871984/ac7265.log)

melihcoban, 2021-07-25:
I am having the same issue in Thinkpad T440p with Intel(R) Dual Band Wireless-N 7260 wifi card. I can use the bluetooth and wifi together but there are stutters and choppy audio with my AirPods and sometimes sound drops randomly. However, when I close the WiFi Airpods works fine with the ethernet cable.

zxystd, 2021-08-02:
@sirius-2 不介意的话麻烦加QQ群(1132224334)帮忙测试ac7265的问题，目前群里没有找到有跟你遇到一样问题的人。

rboldini, 2021-08-03:
With latest Alphas and 7265AC, bluetooth has the same problem.
AirPods and ZK606(low budget speaker ) work but at a ridiculous distance from my laptop, like (for AirPods less than) ~ 4-5cm.
Try to cover proximity sensors on your AirPods and then bring them close to the laptop near the WiFi card, they will ring.

[image]
[image]
[image]

singhalrishi27, 2021-08-07:
Issue is quite similar on wireless ac  7260 As well while Bluetooth and wifi works at the same time but audio gets stuttered Every now and then

williambj1, 2021-10-02:
Chinese discussion moved to QQ, English discussion can be continued in #692

closed 2021-10-02 by williambj1