OpenIntelWireless/itlwm #483: 频繁掉线 fatal firmware error

Basstorm, 2021-01-07:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: latest
- WiFi Card Model: ac9560
- Product ID: a370
- macOS Version: 10.15.7

**Description**

如题，路由器是k2p，之前在宿舍是Redmi AX6 无此问题，放假回来之后家里的是k2p。无大流量传输操作，甚至没怎么动，十分钟之内掉线四五次，掉线后会快速重连。
**Bug Report Archive**
[fatalerror.log](https://github.com/OpenIntelWireless/itlwm/files/5781746/fatalerror.log)

**Kext Download Source**

zxystd, 2021-01-07:
可以看一下你的路由器设置么？

Basstorm, 2021-01-07:
[image]

Creedowl, 2021-01-07:
同一个问题 #477

zxystd, 2021-01-07:
1.1.0的版本就不会是吧？

Basstorm, 2021-01-07:
没试过1.1.0，我去测试一下

zxystd, 2021-01-07:
内测体验群：1132224334

Creedowl, 2021-01-07:
@zxystd 我在最近几个alpha版更新后出现的这个问题，之前一直跟随最新alpha版都是正常的

Basstorm, 2021-01-07:
> 1132224334

找不到这个群

Basstorm, 2021-01-07:
> 1.1.0的版本就不会是吧？

目前看来1.1.0很稳定，没这个情况

williambj1, 2021-01-07:
@Basstorm 尝试一下这个版本

Basstorm, 2021-01-07:
> @Basstorm 尝试一下这个版本 [5cb203e](https://github.com/OpenIntelWireless/itlwm/commit/5cb203e49b2ae5a69cc8453cf4ea510ddb8b0b57)
> 
> [5cb203e.zip](https://github.com/OpenIntelWireless/itlwm/files/5782018/5cb203e.zip)

这个版本也没问题

zxystd, 2021-01-07:
现在可以搜索到群了。

paranoiagu, 2021-01-08:
> 1.1.0的版本就不会是吧？

大概1个月前的1.2.0版本也也不会。

liuzhaobo1999, 2021-01-12:
> > 1.1.0的版本就不会是吧？
> 
> 目前看来1.1.0很稳定，没这个情况

弱弱的问下，是如何做到随意切换版本的？@Basstorm

Basstorm, 2021-01-12:
> > > 1.1.0的版本就不会是吧？
> > 
> > 目前看来1.1.0很稳定，没这个情况
> 
> 弱弱的问下，是如何做到随意切换版本的？@Basstorm

下不同版本efi重启就行了

williambj1, 2021-02-25:
Discussion continued in QQ

closed 2021-02-25 by williambj1

win1010525, 2021-02-25:
QQ is prohibited
[image]