OpenIntelWireless/itlwm #489: 2.4GHz热点无法连接，5GHz正常（intel AX200网卡，M.2/NGFF A+E key）

sungoku, 2021-01-12:
**Have You Read Our Docs*
Yes

**Are You Reporting A Bug**
Yes

**Environment**
- OpenCore0.6.5
- Kext Version: itlwm 1.1.0 stable (Airportitlwm 1.0 beta catalina)
- WiFi Card Model: intel AX200
- Product ID: 0x2723
- macOS Version: 10.15.7 (19H15)

**Description**
你好，刚刚尝试了最新的kext，按照教程推荐的方法安装。连接手机热点的时候发现，5GHz频段的热点可以正常连接，2.4GHz频段的热点却无法连接。
具体表现为：2.4G的热点可以在列表中看到，点击连接后，无论是使用Heliport配合itlwm，还是使用AirportItlwm，均无法成功连接，而且之后在热点列表中，刚才试图连接的2.4GHz热点就很难再刷出来了，开关WIFI都不行，其它热点正常显示在选择列表中。重启不能解决问题。反复连接2.4GHz，一段时间连不上后，Heliport还会自己关闭WIFI。
另外蓝牙在睡眠后出现一次无法使用的情况，还在测试。

非常感谢你们开发的扩展，对你们的付出致敬，期待更加完善。

**Bug Report Archive**
[bugreport_56140.zip](https://github.com/OpenIntelWireless/itlwm/files/5801702/bugreport_56140.zip)

**Kext Download Source**
在github下载的release文件。

williambj1, 2021-01-12:
Firmware crashed during association. CC @zxystd

zxystd, 2021-01-12:
我的ax200测试正常，可能跟有些卡有关系吧，另外你试试1.2.0版本。

sungoku, 2021-01-13:
买的是COMFAST的卡，在Ubuntu20.04上测试了没有问题。
我今天替换成1.2.0的测试版kext试试，稍后汇报情况。

sungoku, 2021-01-13:
迫不及待的进行了初步尝试，使用**1.2.0的Airportitlwm**，手机设置2.4GHz的热点，点击连接，反复出现下图：
[image]

然后把手机热点改成5GHz频段，几秒种后连接成功。
附上检测log
恳请大神查看一下问题在哪

[AirItlDebug_2021-01-13_10-44-23.zip](https://github.com/OpenIntelWireless/itlwm/files/5806075/AirItlDebug_2021-01-13_10-44-23.zip)

稍晚些时间我再试试1.2.0的itlwm，希望能解决。

sungoku, 2021-01-13:
刚刚测试了1.2.0的itlwm，依然如故。。。
不过网速倒是可以在istat菜单显示了。
另外，Heliport能够自动连接已经保存的热点，而Airportitlwm不能，必须自己点选连接。这些都无伤大雅，可以克服。
虽说现在的路由器拥有5G频段也很普遍，但是很多时候依然还是需要连接2.4G的网络啊，比如我的办公网络环境。。。
怎么才能解决2.4G的连接问题呢？如果需要，我可以配合进行调试的。
@zxystd @williambj1

zxystd, 2021-01-13:
你可以修改你的WiFi信道试试

sungoku, 2021-01-14:
@zxystd  修改路由器的2.4G信道还是网卡的？路由器的要进管理界面了，恐怕办公室的不好弄，改网卡的信道在哪里设置？

sungoku, 2021-01-14:
@zxystd 使用手机热点，2.4G频段，把信道从1到11都试了，不行。

sungoku, 2021-01-14:
刚刚看到1.2.0 alpha有更新，是了最新的f657d28，itlwm，还是不可以。

新的Log打包在此。
[bugreport_19370.zip](https://github.com/OpenIntelWireless/itlwm/files/5814534/bugreport_19370.zip)

sungoku, 2021-01-21:
我退掉了COMFAST的网卡，换了另一个牌子的AX200，依然如此。还是兼容性的问题？？？
@zxystd @williambj1

sungoku, 2021-01-22:
各位大佬，可能没人关注我这个issue了。不过我还是更新一下状况吧。
今天在油管看到一个哥们升级WIFI，也是Dell的本子，注意到一个细节，他说原先的kill 1535卡换成intel的卡以后天线就是要反着接，也是黑对黑，白对白。而不是按照main对白，aux对黑的原先的接法。于是我抱着试试的心态对调了一些天线。目前WIFI双频都能连接了，但是蓝牙变成了不可用状态。蓝牙用的也是这里的最新版firmware，不知道为啥。 @zxystd @williambj1

sungoku, 2021-01-22:
补充一下，使用hackintool看，蓝牙驱动是加载了的，看下图。
[image]
@zxystd @williambj1

win1010525, 2021-01-24:
Nothing shows during Bluetooth scan or can't open Bluetooth.

sungoku, 2021-01-29:
@win1010525 Bluetooth get back to function after reboot MacOS. Bluetooth earphone works, but bluetooth mouse did not.
WiFi could connect to 2.4G and 5G ap now, yet the scan was a bit problem ,some netowrk could not show in list, must turn off/on the wifi to get them showing in the list and connect.
I hope this kext gets more steable and more fast in network speed. For now, I am closing this issue. If anyone meet this wired problem in 2.4G or 5G, maybe you connect the antenna the wrong way, don't believe the color or the label in your wifi card, test and reconnect.

closed 2021-01-29 by sungoku

In1t0, 2021-02-03:
我的是更新之后5G频段的没法用了，2.4的正常