OpenIntelWireless/itlwm #491: Hotspot support

MichaelDevC, 2021-01-13:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: v1.2.0-alpha
- WiFi Card Model: AC 8260
- Product ID: 0x24f4
- macOS Version: MacOS BigSur 11.1

**Description**

I can't use iPhone Hotspot... i read in the docs that is not compatible yet, but i'm opening that issue to ask developers to add that feature as soon as possible!
I'm here to test and help developers with Hotspot support development!
Thanks

**Bug Report Archive**

[Report_2021-01-13_08-59-01.log](https://github.com/OpenIntelWireless/itlwm/files/5807211/Report_2021-01-13_08-59-01.log)

**Kext Download Source**

AiportItlwm v1.2.0-alpha#9cc2c4e

lgariv, 2021-01-13:
It works fine for me...

Catalina (latest), AirportItlwm, AC-9462 gen 1 (same firmware as 9560).

Note that for _automatic_ hotspot, you need to have both Bluetooth and WiFi turned on on both of your devices.

Sansyyxd, 2021-01-13:
[image]
Is this the error you also get?

MichaelDevC, 2021-01-13:
That's my setup!

iPhone 11

Bluetooth and Wi-Fi enabled on my hackintosh...

[image]
[image]
[image]

Sansyyxd, 2021-01-13:
yeah having some issue on Catalina I guess it just doesn't have support yet

labrouss, 2021-01-13:
Well there is always the alternative of using the usb cable to share internet ... just saying ...

MichaelDevC, 2021-01-13:
> Well there is always the alternative of using the usb cable to share internet ... just saying ...
> […](#)

Yes... but wireless sharing is better!

I open that issue to help and ask developers to add that feature ;)

MichaelDevC, 2021-01-13:
> yeah having some issue on Catalina I guess it just doesn't have support yet

Also for me on MacOS BigSur

williambj1, 2021-01-16:
Someone posted a solution in Chinese #310, maybe it will help you.

MichaelDevC, 2021-01-16:
Yes, sometimes this works for me, but isn't stable and when i need that (when i'm not at home) is very intricate

williambj1, 2021-01-16:
I guess the best way to solve your issue right now is to use HeliPort with itlwm instead of AirportItlwm before we have a deeper understanding of Instant Hotspot's mechanism.

MichaelDevC, 2021-01-16:
> I guess the best way to solve your issue right now is to use HeliPort with itlwm instead of AirportItlwm before we have a deeper understanding of Instant Hotspot's mechanism.

Thanks I know that!

Leave issue open so that feature can be added in the next v1.3.0 AiportItlwm

MichaelDevC, 2021-02-07:
Can we mark that as a feature request?

williambj1, 2021-02-25:
I guess this could be part of #292

closed 2021-02-25 by williambj1