OpenIntelWireless/itlwm #494: wifi显示正常连接，存在间隔断流问题

groveswallow, 2021-01-15:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
yes

**Environment**

- Kext Version: 1.2.0 stable BigSur
- WiFi Card Model: AX200 003版
- Product ID: 0x2723
- macOS Version: BigSur 11.1

**Description**
wifi能连接，正常使用一段时间后，虽然在设置中显示wifi正常连接，实际已经断流。不做任何操作，隔一段时间，又可上网一段时间，然后又断流，如此重复。
**Bug Report Archive**
[Report_2021-01-15_14-19-23.log](https://github.com/OpenIntelWireless/itlwm/files/5818914/Report_2021-01-15_14-19-23.log)

**Kext Download Source**

williambj1, 2021-01-15:
用 dmesg 重新生成一下日志，谢谢

groveswallow, 2021-01-15:
是这样的日志吗？
[Report.log](https://github.com/OpenIntelWireless/itlwm/files/5819626/Report.log)

groveswallow, 2021-01-15:
> 用 dmesg 重新生成一下日志，谢谢

抱歉，没改boot-args。
[AirItlDebug_2021-01-15_19-07-06.zip](https://github.com/OpenIntelWireless/itlwm/files/5820117/AirItlDebug_2021-01-15_19-07-06.zip)
[dmesg.log](https://github.com/OpenIntelWireless/itlwm/files/5820118/dmesg.log)

zxystd, 2021-01-16:
使用5G频段的Wi-Fi。

groveswallow, 2021-01-16:
> 使用5G频段的Wi-Fi。

非常感谢，试了一下5g频段，暂时没有出现断流的情况。
但是家里路由不支持5g频段，可以有方法解决2.4g频段下的这个问题吗？

AdamYoung66, 2021-01-26:
我AX1650X无论是airport或者heliport 1.3.0跟1.2.0无论是5GHZ还是2.4GHZ都是也有同样的问题，请问airport要怎么生成日志？

williambj1, 2021-01-29:
> 我AX1650X无论是airport或者heliport 1.3.0跟1.2.0无论是5GHZ还是2.4GHZ都是也有同样的问题，请问airport要怎么生成日志？

@AdamYoung66 Instructions can be found in the docs

groveswallow, 2021-01-29:
> 我AX1650X无论是airport或者heliport 1.3.0跟1.2.0无论是5GHZ还是2.4GHZ都是也有同样的问题，请问airport要怎么生成日志？

其实我也是跟你一样的现象，用手机的5G无线热点没有这种情况，但是一旦用路由器的wifi 2.4和5g都会存在断流，5g的恢复时间要短一些。

Kuibagit, 2021-02-08:
修改MTU大小-已解决断流问题
https://github.com/OpenIntelWireless/itlwm/issues/477

groveswallow, 2021-02-08:
> 修改MTU大小-已解决断流问题
> #477

不太明白mtu，请问设置成多少合适呢？

win1010525, 2021-02-08:
As a reference, you can see your item log. If it shows `OUTPUT Error` with code 131, you need to set MTU lower.

win1010525, 2021-02-08:
> As a reference, you can see your item log. If it shows `OUTPUT Error` with code 131, you need to set MTU lower.

At least equal to your router setting.

groveswallow, 2021-02-08:
> > As a reference, you can see your item log. If it shows `OUTPUT Error` with code 131, you need to set MTU lower.
> 
> At least equal to your router setting.

This is my log.
[Report_2021-02-08_19-22-46.log](https://github.com/OpenIntelWireless/itlwm/files/5943411/Report_2021-02-08_19-22-46.log)
It seems not have the code 131.What should I do to solve this question？
Thanks!

Kuibagit, 2021-02-08:
> > 修改MTU大小-已解决断流问题
> > #477
> 
> 不太明白mtu，请问设置成多少合适呢？

不同的网卡不同的系统MTU值应该不一样吧。我的是AX201，Catalina 10.15.7设置了1482就正常了。

groveswallow, 2021-02-08:
> > > 修改MTU大小-已解决断流问题
> > > #477
> > 
> > 
> > 不太明白mtu，请问设置成多少合适呢？
> 
> 不同的网卡不同的系统MTU值应该不一样吧。我的是AX201，Catalina 10.15.7设置了1482就正常了。

谢谢。是不是路由器中的mtu无需修改，只用修改网卡的mtu进行测试？

win1010525, 2021-02-26:
Can you guys have a test on the latest 1.3.0-alpha？I think it have been fixed.

CodeApePro, 2021-04-22:
@zxystd @win1010525 您好,我这边也遇到了疑似WiFi断流的情况,具体表现为,我有一个网页有一条websocket连接(有心跳),在我打开一个网页放在那不动的情况下,大概一两个小时之后总会连接被断开,期间没有发现wifi断开重连的情况.其他系统下没有发现这个问题.大佬可以了解和参考下.

zxystd, 2021-05-07:
Try the latest one: v2.0.0-alpha?

CodeApePro, 2021-05-07:
@zxystd 好的 我试试.

williambj1, 2021-10-02:
Feel free to post comments if this issue still occurs in the latest release, providing log files together will be much appreciated.

closed 2021-10-02 by williambj1