OpenIntelWireless/itlwm #495: Request: Disabled 2.4GHz Support and or Force 5GHz WiFi

alinach120602, 2021-01-15:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: 1.2.0 
- WiFi Card Model: Intel AX200
- Product ID: AX200
- macOS Version: 11.1

**Description**
This is a feature request, to disable 2.4GHz WiFi connections either via a tool or the driver itself and or Force 5GHz Connections to avoid Bluetooth Interference. As I'd like to have Bluetooth 5.0 Devices connected and use WiFi. and there's also the problem that usually when connection to a Router with Band Steering from 2.4 to 5.8 and vice-versa the WiFi Connection drops with AirportItlwm. I'd just like to always use 5GHz WiFi on any Network and be able to use Bluetooth because of it. I use a few Acces Points with the Same SSID for 2.4 and 5GHz and have no way of configuring the Router

**Bug Report Archive**

Not needed for this type of Issue

**Kext Download Source**

2.1.0 Release GitHub

zxystd, 2021-01-16:
do you mean disable ROAM?

lunjielee, 2021-01-16:
@alinach120602 FYI: [How do you force a 5 GHz wifi connection?](https://apple.stackexchange.com/questions/131852/how-do-you-force-a-5-ghz-wifi-connection)

alinach120602, 2021-01-16:
> do you mean disable ROAM?

Yes, I guess I mean ROAM

alinach120602, 2021-01-16:
> @alinach120602 FYI: [How do you force a 5 GHz wifi connection?](https://apple.stackexchange.com/questions/131852/how-do-you-force-a-5-ghz-wifi-connection)

That doesn't work btw whatever 5GHz channel I force it won't work

win1010525, 2021-01-17:
Why not just connecting to 5GHz Wi-Fi? I think it is insignificant to disable it, and many people are still using 2.4GHz Wi-Fi now.

satyrnhub, 2021-01-17:
> Why not just connecting to 5GHz Wi-Fi? I think it is insignificant to disable it, and many people are still using 2.4GHz Wi-Fi now.

Some people (such as myself) have a Wi-Fi network with both 2.4GHz and 5GHz, and it would be nice to have an option to force a 5GHz channel.

win1010525, 2021-01-17:
Oh I see.

1alessandro1, 2021-01-17:
What about adding a boolean property in `itlwm.kext` `Info.plist`file next to SSID and password fields such as `5GHz only` True/False?

alinach120602, 2021-01-17:
> What about adding a boolean property in `itlwm.kext` `Info.plist`file next to SSID and password fields such as `5GHz only` True/False?
Or something like a itlwm_freq_only=5G Nvram Variable

satyrnhub, 2021-01-17:
> What about adding a boolean property in `itlwm.kext` `Info.plist`file next to SSID and password fields such as `5GHz only` True/False?

Good idea. I don't know much actual programming, but I might try to get it to work...

satyrnhub, 2021-01-17:
Threshold values look hardcoded, but you can probably just change them so...

satyrnhub, 2021-01-17:
I made a fork which disables band steering: https://github.com/satyrnhub/itlwm

Pretty easy to do.

1alessandro1, 2021-01-18:
@satyrnhub, what about making a PR later to add your work as an option (and not forced?)

@zxystd could this be even possible with a boolean in the info.plist?

zxystd, 2021-01-18:
@1alessandro1 I am wondering weather there is another good way to escape reboot and configure the info.plist.

satyrnhub, 2021-01-18:
> @satyrnhub, what about making a PR later to add your work as an option (and not forced?)
> 
> @zxystd could this be even possible with a boolean in the info.plist?

I have little knowledge with actual programming, the fact that it works is a surprise lol. If someone else wants to do that, be free to do so. All I did was remove the bit that chooses whether `selbs` should be `selbs5` (5GHz) and `selbs2` (2.4GHz). I imagine making that toggle-able wouldn't be too hard.

williambj1, 2021-01-18:
This issue is a duplication of #487

itlwm uses an algorithm to determine whether 2.4GHz or 5GHz should be used. If the RSSI value, simpler knows as signal strength, is lower than the threshold value given by OpenBSD, it will cause instabilities other issues. Simply disabling 2.4GHz is impractical and violates IEEE 802.11 standards. Feel free to modify the code for your own needs but you are discouraged to publish it under the same name.

@satyrnhub I can imagine having an option to force connect to a specific BSSID, but note that none of those solutions including your proposal are easily toggleable in the Airport version.

1alessandro1, 2021-01-19:
> Simply disabling 2.4GHz is impractical and violates IEEE 802.11 standards.

How can we choose the channel which we prefer then?

1alessandro1, 2021-01-19:
> Simply disabling 2.4GHz is impractical and violates IEEE 802.11 standards.

Are there any other solutions to allow the users (which can disable `band steering`) to force 5GHz mode with a quirk/nvram variable?

win1010525, 2021-01-19:
It always automatically connects 5GHz Wi-Fi, if it connects to 2.4GHz Wi-Fi, the only explanation is the RSSI of your 5GHz Wi-Fi is unstable (may also caused by itlwm's unstability ).

williambj1, 2021-01-19:
> > Simply disabling 2.4GHz is impractical and violates IEEE 802.11 standards.
> 
> Are there any other solutions to allow the users (which can disable `band steering`) to force 5GHz mode with a quirk/nvram variable?

We won't consider those hacks but specifying BSSID is reasonable.

satyrnhub, 2021-01-19:
> > > Simply disabling 2.4GHz is impractical and violates IEEE 802.11 standards.
> > 
> > 
> > Are there any other solutions to allow the users (which can disable `band steering`) to force 5GHz mode with a quirk/nvram variable?
> 
> We won't consider those hacks but specifying BSSID is reasonable.

Maybe even just prioritizing 5GHz even more than 2.4GHz.

1alessandro1, 2021-02-03:
> @1alessandro1 I am wondering whether there is another good way to escape reboot and configure the info.plist.

@zxystd is the NVRAM variable a viable option?

win1010525, 2021-02-03:
> > @1alessandro1 I am wondering whether there is another good way to escape reboot and configure the info.plist.
> 
> @zxystd is the NVRAM variable a viable option?

As bat.bat said, this is a wontfix issue  because you can easily fix this by changing SSID. Simply disable 2.4G will cause instabilities and violates IEEE 802.11 standards.

win1010525, 2021-02-03:
Shouldn’t you close this issue? @williambj1

XEmpty, 2021-05-03:
Hi, simple question. It's maybe possible to connect with selected BSSID? I mean I don't have access to router and I need in all my system (linux, windows) select manual specific BSSID . Is there any possible way to get that right? info.plist or in HeliPort ? ( I just need to know where enter/put specific BSSID (because right now it's automatic = as best signal) )
Thanks.

closed 2021-08-13 by alinach120602