OpenIntelWireless/itlwm #499: [AirportItlwm]AC7265 能够驱动，但无法搜索到任何 WIFI

ciallowo, 2021-01-18:
**Have You Read Our Docs**

**Yes**
**Are You Reporting A Bug**

**Yes**
**Environment**

- Kext Version: 1.2.0 stable
- WiFi Card Model: Intel Dual Band Wireless **AC 7265**
- Product ID: 0x8086, 0x5010
- macOS Version: Big.Sur.11.1.0.20C69

**Description**

**AC7265 能够驱动**，“关于本机”里面查看是完全驱动了的。Airportitlwm 各版本我都已经试过一边了，目前在用 1.2.0 stable 但无法搜索到任何 WIFI。
[image]
**Bug Report Archive**

[AirItlDebug.zip](https://github.com/OpenIntelWireless/itlwm/files/5830610/AirItlDebug.zip)
**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/download/v1.2.0/AirportItlwm_v1.2.0_stable_BigSur.kext.zip

win1010525, 2021-01-18:
No itlwm logs. Maybe crashed? @zxystd

williambj1, 2021-01-18:
@ciallowo 尝试重新生成一下日志，给出的zip里没抓到

win1010525, 2021-01-18:
Well your log isn't using the newest debugging file released 37 minutes ago. Please try the new one which included a check about your nvram settings.

win1010525, 2021-01-18:
Here's the file.
[AirItlDebug-Big-Sur.sh.zip](https://github.com/OpenIntelWireless/itlwm/files/5831025/AirItlDebug-Big-Sur.sh.zip)

ciallowo, 2021-01-18:
@williambj1 @win1010525 I'll do it again.

ciallowo, 2021-01-18:
Here is the debug log file, and I extract real time Airport logs about 60 seconds.
[AirItlDebug.zip](https://github.com/OpenIntelWireless/itlwm/files/5831118/AirItlDebug.zip)

win1010525, 2021-01-18:
Still no itlwm logs. @zxystd

ciallowo, 2021-01-19:
> Still no itlwm logs. @zxystd

Hmmm...

What else can I provide to help contributors to solve this problem?

williambj1, 2021-01-19:
try resetting your network preferences and use `itlwm.kext` with HeliPort

win1010525, 2021-01-19:
This maybe insignificant as he has already loaded airportitlwm successfully but no logs were coming out and nothing happened. @williambj1

win1010525, 2021-01-19:
If errors in bootloader exist, the kext won't load successfully.

williambj1, 2021-01-19:
Can't tell what's wrong without the logs. Maybe consider using dmesg manually to extract the logs, instructions can be found in our docs.

AirDronzer, 2021-01-23:
@ciallowo Found any Fix I m also facing the same issue With AC7265 Aiportitlwm

ciallowo, 2021-01-24:
> @ciallowo Found any Fix I m also facing the same issue With AC7265 Aiportitlwm

Not yet. Maybe you can try to provide some log files.

win1010525, 2021-02-03:
Can you get a log with new debug files?

HsunGong, 2021-02-12:
Same error after update to bigsur11.2(or 11.0).
Use itlwm+heliport, works fine(can find wifi and get connected).
However, if use airportitlwm, it just can not find any wifi, and what's worse is that I can not switch wifi button to open.

williambj1, 2021-02-25:
Will reopen if valid logs are provided

closed 2021-02-25 by williambj1