OpenIntelWireless/itlwm #500: itlwmx下AX201网卡经常断流

wiseideal, 2021-01-19:
**Have You Read Our Docs**

yes

**Are You Reporting A Bug**

yes

**Environment**

- Kext Version: 1.0.0
- WiFi Card Model: AX201
- Product ID: intel
- macOS Version: 11.1

**Description**

下载后放入oc中 可以搜到wifi热点 连接可以上网 但是时不时会断流

**Bug Report Archive**

HeliPort中没有发现此按钮 点击后无效果 状态栏出现wifi图标 仅此

[Log_2021-01-19_17-59-55.log](https://github.com/OpenIntelWireless/itlwm/files/5835288/Log_2021-01-19_17-59-55.log)
**Kext Download Source**

github release

我也试过itlwm 1.2.0版本 问题依旧 HeliPort直接提示没有运行

wiseideal, 2021-01-19:
[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/5835325/wifi.log)
找到了wifi log 希望有所帮助

williambj1, 2021-01-19:
版本太老了，用新版本的时候请确保驱动注入正确

closed 2021-01-19 by williambj1