OpenIntelWireless/itlwm #502: kernel panic

anonymous5l, 2021-01-21:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v1.2.0 stable
- WiFi Card Model: AC3165
- Product ID: 3165D2W
- macOS Version: 10.14.6

**Description**

*** Panic Report ***
panic(cpu 0 caller 0xffffff7f865694cd): itlwm: nic locks counter 0
Backtrace (CPU 0), Frame : Return Address
0xffffff8129d5b750 : 0xffffff80033ae5ad mach_kernel : _handle_debugger_trap + 0x47d
0xffffff8129d5b7a0 : 0xffffff80034ea1f5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff8129d5b7e0 : 0xffffff80034db92a mach_kernel : _kernel_trap + 0x50a
0xffffff8129d5b850 : 0xffffff800335b9d0 mach_kernel : _return_from_trap + 0xe0
0xffffff8129d5b870 : 0xffffff80033adfc7 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff8129d5b990 : 0xffffff80033ade13 mach_kernel : _panic + 0x63
0xffffff8129d5ba00 : 0xffffff7f865694cd com.zxystd.itlwm : __ZN6ItlIwm21iwm_nic_assert_lockedEP9iwm_softc + 0xad
0xffffff8129d5ba30 : 0xffffff7f86569726 com.zxystd.itlwm : __ZN6ItlIwm12iwm_nic_lockEP9iwm_softc + 0x36
0xffffff8129d5ba70 : 0xffffff7f86510d4b com.zxystd.itlwm : __ZN6ItlIwm12iwm_send_cmdEP9iwm_softcP12iwm_host_cmd + 0x57b
0xffffff8129d5bb60 : 0xffffff7f865116e5 com.zxystd.itlwm : __ZN6ItlIwm16iwm_send_cmd_pduEP9iwm_softcjjtPKv + 0x135
0xffffff8129d5bc10 : 0xffffff7f8655cd5b com.zxystd.itlwm : __ZN6ItlIwm13iwm_sf_configEP9iwm_softci + 0x16b
0xffffff8129d5bcc0 : 0xffffff7f8655847f com.zxystd.itlwm : __ZN6ItlIwm22iwm_run_init_mvm_ucodeEP9iwm_softci + 0x20f
0xffffff8129d5bd20 : 0xffffff7f8655ea7b com.zxystd.itlwm : __ZN6ItlIwm11iwm_init_hwEP9iwm_softc + 0xbb
0xffffff8129d5bd80 : 0xffffff7f8655e14f com.zxystd.itlwm : __ZN6ItlIwm8iwm_initEP6_ifnet + 0x8f
0xffffff8129d5bdf0 : 0xffffff7f865627d0 com.zxystd.itlwm : __ZN6ItlIwm13iwm_init_taskEPv + 0x110
0xffffff8129d5be40 : 0xffffff7f8654d7ff com.zxystd.itlwm : __Z12taskq_threadPv + 0x6f
0xffffff8129d5bfa0 : 0xffffff800335b0ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.itlwm(1.2)[A27F48C4-5E37-3474-BFD2-34B08F609579]@0xffffff7f864cf000->0xffffff7f871f7fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[8C82DA30-EF91-3207-8D57-331C46E49D31]@0xffffff7f8401f000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[2328355E-F574-3D2A-A77D-A69917B3ED1D]@0xffffff7f83c95000

BSD process name corresponding to current thread: kernel_task
Boot args: -v debug=0x100 keepsyms=1 igfxonln=1

Mac OS version:
18G7016

Kernel version:
Darwin Kernel Version 18.7.0: Tue Nov 10 00:07:31 PST 2020; root:xnu-4903.278.51~1/RELEASE_X86_64
Kernel UUID: 0F94ABBC-EB06-3DEE-A01B-6F82102C1848
Kernel slide:     0x0000000003000000
Kernel text base: 0xffffff8003200000
__HIB  text base: 0xffffff8003100000
System model name: MacBook9,1 (Mac-9AE82516C7C6B903)

System uptime in nanoseconds: 19822820652

EOF
Model: MacBook9,1, BootROM 429.60.3.0.0, 2 processors, Intel Core m5, 1.6 GHz, 8 GB, SMC 2.35f108
Graphics: Intel UHD Graphics 615, Intel UHD Graphics 615, spdisplays_builtin
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, LPDDR3, 1600 MHz, SK Hynix, -
Bluetooth: Version 6.0.14d12, 3 services, 26 devices, 1 incoming serial ports
Network Service: Wi-Fi, Ethernet, en0
PCI Card: pci8086,9d71, Audio Device, internal
USB Device: USB 3.0 Bus
USB Device: 4-Port USB 3.0 Hub
USB Device: Arduino Leonardo
USB Device: Magic Trackpad 2
USB Device: Miscellaneous Device
USB Device: USB2.0 Hub
USB Device: Bluetooth HCI
Thunderbolt Bus: 

**Bug Report Archive**

**Kext Download Source**

williambj1, 2021-01-21:
Duplicate of #480

closed 2021-01-21 by williambj1

anonymous5l, 2021-01-21:
panic on boot up, not from the wake up.