OpenIntelWireless/itlwm #505: no variant specified, falling back to release

jaigup, 2021-01-25:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Maybe

**Environment**

- Kext Version: 1.2
- WiFi Card Model: Intel 8260NGW
- Product ID: 806721-001
- macOS Version: 11.1

**Description**
Adding itlwm.kext and adding to the config.plist does not lead to a wifi icon at the top right when trying to install. I ran `kextstat | grep -i "itlwm"` and got 
`no variant specified, falling back to release`
`   71     0 0                      0xd37000      0xd37000      com.zxystd.itlwm (1.2.0) A27F48C4-5E37-3474-BFD2-34B08F609579 <46 14 6 5 3 1>`

**Bug Report Archive**
not in full macOS, just recovery | N/A

**Kext Download Source**
github releases

ErrorErrorError, 2021-01-25:
> not in full macOS, just recovery | N/A

Using `itlwm.kext` in installer will NOT show you the Wi-Fi icon since `itlwm.kext` emulates as an ethernet connection. You can use `AirportItlwm.kext` so the Wi-Fi icon shows on the top right during installation.

closed 2021-01-25 by jaigup

jaigup, 2021-01-25:
Yea I figured it out. `AirportItlwm.kext` solved the problem