OpenIntelWireless/itlwm #508: AX 201 Wifi

itendtostare, 2021-01-27:
**Have You Read Our Docs**
 Yes

**Are You Reporting A Bug**
 Yes

**Environment**

- Kext Version: v1.2.0 stable / v1.3.0-alpha
- WiFi Card Model: Intel AX201
- Product ID: PciRoot(0x0)/Pci(0x14,0x3)
- macOS Version: Big Sur 11.2

**Description**
 Wifi is not visible in network preferences / does not find any wifi in the menu. Wifi is not active. Kext loads.

**Bug Report Archive**
 
```----- Kext Status -----

   69    0 0                  0xd47000   0xd47000   com.zxystd.AirportItlwm (1.3.0) 0D4F0B54-7D16-3AE3-9B68-2A73D06DB064 <68 46 14 6 5 3 1>

----- Airport Info -----

----- Airport Sys Report -----

Wi-Fi:

      Software Versions:
          CoreWLAN: 16.0 (1657)
          CoreWLANKit: 16.0 (1657)
          Menu Extra: 17.0 (1728)
          System Information: 15.0 (1502)
          IO80211 Family: 12.0 (1200.12.2b1)
          Diagnostics: 11.0 (1163)
          AirPort Utility: 6.3.9 (639.15)

----- Bluetooth Sys Report -----

Bluetooth:

      Apple Bluetooth Software Version: 8.0.3d9
      Hardware, Features, and Settings:
          Bluetooth Low Energy Supported: Yes
          Handoff Supported: No
          Instant Hot Spot Supported: No
          Manufacturer: Intel
          Transport: USB
          Firmware Version: 10.256 (10.256)
          Connectable: Yes
          Auto Seek Pointing: On
          Remote wake: Off
          Vendor ID: 0x8087
          Product ID: 0x0026
          Auto Seek Keyboard: On
      Devices (Paired, Configured, etc.):
          Device 0:
              Major Type: Peripheral
              Minor Type: Trackpad
              Services: Magic Trackpad 2
              Manufacturer: Broadcom (0x5, 0x240C)
              Battery Level: 53%
              Bluetooth Core Spec: 3.0
              Firmware Version: 0x0102
              Vendor ID: 0x004C
              Product ID: 0x0265
              Host Connectable: Yes
              EDR Supported: Yes
              eSCO Supported: No
              SSP Supported: Yes
[... 179 lines cut ...]
2021-01-26 00:01:25.855685+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] _hashesNeedUpdating called. forceRebuild: false
2021-01-26 00:01:25.855717+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] updateHashes called
2021-01-26 00:01:25.858932+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] Hash store should rebuild at: 2021-02-09 11:10:58 +0000, lastMod: 2021-01-25 11:10:58 +0000. timeToNextUpdate: 1249772sec
2021-01-26 00:01:25.859104+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] regenerateEntireDB = false
2021-01-26 00:01:25.859130+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] Change anchor: <CNChangeHistoryAnchor: 0x7fb71ba10900: token=<NSPersistentHistoryToken - {
2021-01-26 00:01:25.859140+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] load called twice on SDAirDropHashStoreCDB
2021-01-26 00:01:25.859152+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] Finished enumerateContactChangesWithKeys.
2021-01-26 00:01:25.859160+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] schedulePersist - Nothing to persist.
2021-01-26 00:01:25.859160+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] updateHashes finished.
2021-01-26 00:01:25.859166+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] _hashesNeedUpdating called. forceRebuild: false
2021-01-26 00:01:25.859167+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] updateHashes called
2021-01-26 00:01:25.862234+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] Hash store should rebuild at: 2021-02-09 11:10:58 +0000, lastMod: 2021-01-25 11:10:58 +0000. timeToNextUpdate: 1249772sec
2021-01-26 00:01:25.862400+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] regenerateEntireDB = false
2021-01-26 00:01:25.862426+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] Change anchor: <CNChangeHistoryAnchor: 0x7fb71ba23330: token=<NSPersistentHistoryToken - {
2021-01-26 00:01:25.862429+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] load called twice on SDAirDropHashStoreCDB
2021-01-26 00:01:25.862440+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] Finished enumerateContactChangesWithKeys.
2021-01-26 00:01:25.862448+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] schedulePersist - Nothing to persist.
2021-01-26 00:01:25.862449+0000 0x6db8     Default     0x0                  465    3    sharingd: [com.apple.sharing:SDAirDropHashStore] updateHashes finished.
2021-01-26 00:02:04.683654+0000 0x11a0     Info        0x0                  465    0    sharingd: [com.apple.sharing:Daemon] XPC_ERROR_CONNECTION_INVALID <0x7fb709c10dc0>
```

**Logs**

**Kext Download Source**
 https://github.com/OpenIntelWireless/itlwm/releases

williambj1, 2021-01-29:
Follow the instructions in our docs and reset your network preferences.

closed 2021-01-29 by williambj1

itendtostare, 2021-01-29:
I did all these actions and I still have the same issue, I used the resetting [system preferences troubleshooting here](https://openintelwireless.github.io/itlwm/Troubleshooting.html#kernel-extension-loading-status)

Reset network preferences:

[image]

After reboot:

[image]

williambj1, 2021-01-30:
Please attach logs generated by AirItlDebug.sh

M69W, 2021-01-30:
> Wifi is not visible in network preferences / does not find any wifi in the menu. Wifi is not active. Kext loads.

use itlwm.kext + heliport.app or just use airportitlwm.kext ?

itendtostare, 2021-02-01:
@williambj1 here it is, I noticed another user had the same issue recently, maybe it has something to do with the macOS update. When getting the logs, I get a message that says ' Unable to retrieve AirPort interface list ' 
[AirItlDebug_2021-02-01_10-26-50.zip](https://github.com/OpenIntelWireless/itlwm/files/5903363/AirItlDebug_2021-02-01_10-26-50.zip)

williambj1, 2021-02-01:
@itendtostare Hmmm, not a single line of itlwm logs and airport still remains uninitialized. I'm pretty sure this is a configuration issue...

itendtostare, 2021-02-01:
Most probably you are right, I m just not good enough to understand what I would have to try and do, let me know if you have any brilliant idea, or basically should I just wipe HD and do a fresh install and see ? 

I can't even create a new configuration for a wifi from preferences, not sure if that helps. Maybe I can import a configuration from someone else and see if that solves it?

[image]