OpenIntelWireless/itlwm #512: Kext is loaded correctly, but Wi Fi cannot be searched

lovelyxmj, 2021-01-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v1.2.0
- WiFi Card Model: Intel(R) Dual Band Wireless AC 3165
- Product ID: I`m not sure,but I know it`s Intel(R) Dual Band Wireless AC 3165
- macOS Version: 10.15.7

**Description**
Chinese:我已经加载了AirportItlwm,并且在系统内看它似乎已正确加载,但是wifi列表里搜索不到任何Wi-Fi
English:I have loaded airportitlwm, and it seems to have loaded correctly in the system, but no Wi Fi can be found in the WiFi list
A picture:
[image]

**Bug Report Archive**
[AirItlDebug_2021-01-30_14-54-09.zip](https://github.com/OpenIntelWireless/itlwm/files/5897517/AirItlDebug_2021-01-30_14-54-09.zip)

**Kext Download Source**
I download it from github
从Github

williambj1, 2021-01-30:
日志中显示 Airport/CoreWLAN 都没有加载

1. 尝试重置端口
2. 尝试使用 heliport + itlwm

lovelyxmj, 2021-01-30:
How can I reset port?
我应该如何重置端口

lovelyxmj, 2021-01-30:
我使用heilport+itlwm会导致heilport点击打开Wi-Fi没有反应
If I use heilport and itlwm,I will can`t open Wi-Fi in heilport

ArzeLiao, 2021-02-01:
有qq群吗?不知道为什么我的网速有点慢, 下载速度只有2mb/s

lovelyxmj, 2021-02-03:
Can anyone help me?

win1010525, 2021-02-03:
> Can anyone help me?

We wanted to help, but this is so rare that only three people reported this bug with different cards.
All logs of this issue are different. And your log doesn't contain itlwm log. Sorry but I can't help to deal with it.

win1010525, 2021-02-03:
You can try with dmesg

win1010525, 2021-02-03:
@zxystd Three people have reported this issue since I told you last time. If you have time, please check the part of firmware load of iwm.
I've checked their EFI, there have msgbuf=1048576 as well. Still no log.

win1010525, 2021-02-03:
> 有qq群吗?不知道为什么我的网速有点慢, 下载速度只有2mb/s

Off topic. @williambj1

williambj1, 2021-03-15:
Dual boot / Fast startup / Invalid ASPM related issue. Can't do much with this kext, please fix those configuration issues yourself.

closed 2021-03-15 by williambj1