OpenIntelWireless/itlwm #518: ax200网络不稳定，频繁闪断

lcg2015, 2021-02-03:
intel ax200
同一台机器，win10下没有问题，mac下速度下降到win10的1/10,使用pip等安装稍大的包经常超时，怀疑频繁断开连接。

[Log_2021-02-03_14-46-49.log](https://github.com/OpenIntelWireless/itlwm/files/5916624/Log_2021-02-03_14-46-49.log)

dawn-cgj, 2021-02-03:
我也是AX200，速率能到30M左右，大概十分钟断一次，需要开关一下WiFi

win1010525, 2021-02-03:
Invalid

win1010525, 2021-02-03:
https://docs.oiw.workers.dev/itlwm/FAQ.html

win1010525, 2021-02-03:
Performance sucks has been written down in doc.

mahoshojoHCG, 2021-02-03:
我用的也是AX200，之前用的1.2.0的正式版。遇到这个问题之后试着更新使用1.3.0的测试版，问题似乎就解决了。

shasen1235, 2021-02-03:
Probably Big Sur is the one to blame. I have disconnect, DHCP refresh problems and so on.
But my same AX200 works perfectly fine on Catalina and Mojave.

MicalShow, 2021-02-04:
同AX200，网速很猛，最新的测试版基本能跑满200m宽带了，峰值能到233m，就是断流有点频繁，希望修复

hexiosr, 2021-03-03:
ax201,也有这个问题

M69W, 2021-03-03:
建议把系统版本，哪个驱动以及版本等信息描述一下

hexiosr, 2021-03-04:
> 建议把系统版本，哪个驱动以及版本等信息描述一下
电脑：xiaoxin 13 pro 2020
网卡：AX201
系统版本：bigsur11.2.2 
驱动：AirportItlwm v1.3.0 alpha
大概10分钟内会断开网络，但是WiFi还处于连接状态

thep0y, 2021-03-23:
> 我用的也是AX200，之前用的1.2.0的正式版。遇到这个问题之后试着更新使用1.3.0的测试版，问题似乎就解决了。

请问你在上传文件时速度正常马？我这边用speedtest跑的时候上行是30M，但是真正传文件时，只有几十K的速度

thep0y, 2021-03-23:
> 同AX200，网速很猛，最新的测试版基本能跑满200m宽带了，峰值能到233m，就是断流有点频繁，希望修复

你好，我也同ax200，300M的宽带只能跑到130M左右，请问你用的是哪个版本的驱动，用的系统版本是什么

williambj1, 2021-10-02:
Feel free to post comments if the same behavior still occurs in the latest release.

closed 2021-10-02 by williambj1