OpenIntelWireless/itlwm #519: Itlwm does not work after sleep

lilov-zz, 2021-02-03:
I am using Mojave 10.14.6 on an HP Elitebook 830 G5. The wifi card is Intel Ac 8265. I am using latest kext 1.2.0. and Heliport.
The issue is after sleep Heliport WiFi icon is empty and turn wifi on does not work. Same thing happens when I manually turn WIFi off, there is no way to put it manually back on.  Only restart fixes it.
I have read previous threads and tried Itlwmx from previous issues but it does not work.

lilov-zz, 2021-02-03:
[log.txt](https://github.com/OpenIntelWireless/itlwm/files/5920614/log.txt)

williambj1, 2021-02-03:
No itlwm logs. Please follow the instructions in the docs.

lilov-zz, 2021-02-04:
Trying to export log from Heliport but I get this. I am sorry, I am new to this can you tell me how to get the log out?
[image]

lilov-zz, 2021-02-04:
Trying to export log from Heliport but I get this. I am new to this can you help how to bypass this error
[image]

lilov-zz, 2021-02-04:
I think I have managed to do it :-)
[Log_2021-02-04_19-38-42.log](https://github.com/OpenIntelWireless/itlwm/files/5927524/Log_2021-02-04_19-38-42.log)

williambj1, 2021-02-05:
Try toggling wifi off and on

lilov-zz, 2021-02-05:
Same thing, been trying this for the last two days.

Sent from a mobile device. Please excuse typos and formatting.

lilov-zz, 2021-02-06:
Problem is sleep, I can turn wifi on and off when working normally but after sleep Itlwm does not wake.

williambj1, 2021-02-10:
Did you add `msgbuf=1048576` to boot-args? A lot of logs are missing.

lilov-zz, 2021-02-10:
Hi, you are correct. I have now added it and enclose the updated log taken after sleep and being unable to turn wifi ON.
[Log_2021-02-10_15-17-10.log](https://github.com/OpenIntelWireless/itlwm/files/5958435/Log_2021-02-10_15-17-10.log)
If i can do anything else let me know.

Qiangong2, 2021-02-14:
I'm having the same issue on big sur with my AC 9462

win1010525, 2021-02-26:
@williambj1 have you seen the latest provided log?

afzl-wtu, 2021-02-28:
working fine on intel 6205 except i have to turn of and then On it, and it connects again

lilov-zz, 2021-02-28:
I do not have the option to turn it off, after sleep Heliport state/ itlwm is OFF and I can only try to turn it on, but with no success regardless of how many tries I attempt.

zxystd, 2021-03-07:
can you try the latest alpha release.

lilov-zz, 2021-03-07:
Same thing, enclosed is the new log with the alpha 1.30
[Log_2021-03-07_20-15-57.log](https://github.com/OpenIntelWireless/itlwm/files/6097721/Log_2021-03-07_20-15-57.log)

williambj1, 2021-03-15:
Looks like your card is operating with PCI ASPM L1 mode, itlwm does not support ASPM ATM. Make sure you did not inject invalid ASPM values in device properties / _DSM, etc.

lilov-zz, 2021-03-15:
I am afraid you got me here. Do not know what to do and no nelpful Google search found. If not too much trouble can you tell me what and where to look for ? Thanks.

GGorAA, 2021-03-16:
same issue on HP EliteBook 840 G3 with Intel Wireless 8260

hectorox, 2021-03-16:
same here with Intel Wireless-N 7260

adeg, 2021-03-18:
I have the same issue with Intel® Wi-Fi 6 AX201. Using the alpha AirportItlwm-Big Sur build from 2021-03-10.

zeyugao, 2021-03-21:
Same issue with AC 8265 in Big Sur using AirportItlwm

[Report_2021-03-21_19-07-05.log](https://github.com/OpenIntelWireless/itlwm/files/6177271/Report_2021-03-21_19-07-05.log)

Nels2, 2021-03-23:
same as hectorox

ubun222, 2021-04-05:
nuc8在关闭或者睡眠后AirPlay不能用然后会断连，开机第一次连接上Wi-Fi是没问题的

GGorAA, 2021-04-09:
UPDATE: latest 1.3.0 fixed dis issue, now WiFi works even after going to sleep

lilov-zz, 2021-04-10:
@GGorAA Hi, you mean the 1.3.0 alpha? I am running it but nothing changes. Are you using any specific options?

GGorAA, 2021-04-14:
> @GGorAA Hi, you mean the 1.3.0 alpha? I am running it but nothing changes. Are you using any specific options?

Yes, i mean 1.3.0. And also i'm not using any specific options other than Airportitlwm explicitly needs (everything is in the docs)

karelo911, 2021-08-07:
Same problem on same card. Tried itlwm version 2 alpha. I can't turn on wifi after I turn it off, either manually or through sleep.
[bugreport_30678.zip](https://github.com/OpenIntelWireless/itlwm/files/6948614/bugreport_30678.zip)

williambj1, 2021-10-02:
@lilov-zz Hi! Have you tried the latest alpha? What about the ASPM related configs? Thanks.

@karelo911 The log file does not contain any errors or signs of sleeping / toggling on and off.

SesshomaruVN, 2021-11-30:
Hi, I am using AX 201 wifi card. My problem as the same this thread issue. I have read some thread issue at Issuses tab. And try to config aspm = 0 for my wifi card in devices tree. But, wifi still not working after sleep.
I am using this kext [AirportItlwm-Big_Sur-v2.1.0-DEBUG-alpha-3ee6b6a.zip](https://github.com/OpenIntelWireless/itlwm/releases/download/v2.1.0-alpha/AirportItlwm-Big_Sur-v2.1.0-DEBUG-alpha-3ee6b6a.zip)
[image]
This is aspm config.
And here is the log of itlwm after boot (wifi work well}
[Log_2021-12-01_06-30-08.log](https://github.com/OpenIntelWireless/itlwm/files/7629565/Log_2021-12-01_06-30-08.log) 
And here is the log after sleep (can't find any network name)
[Log_2021-12-01_06-31-16.log](https://github.com/OpenIntelWireless/itlwm/files/7629566/Log_2021-12-01_06-31-16.log)
Wifi will work after restart.
Please help me to fix. Thanks so much.

zxystd, 2021-12-01:
[Big Sur.zip](https://github.com/OpenIntelWireless/itlwm/files/7632046/Big.Sur.zip)
@SesshomaruVN your system disable the network card when sleep, so it will never work after wakeup, I have done something in the kext to try to enable the network card again, please try.

SesshomaruVN, 2021-12-01:
> [Big Sur.zip](https://github.com/OpenIntelWireless/itlwm/files/7632046/Big.Sur.zip)
> @SesshomaruVN your system disable the network card when sleep, so it will never work after wakeup, I have done something in the kext to try to enable the network card again, please try.

Thank you so much for helping me.
I have used your kext file above, but It is still not working after sleep.
After inject your kext file, the only one time I can wake from sleep with wifi working (but lost the current wifi in the list) - sorry because of no log file in this case.
After that time, still not working. Here is my log after used above kext.
[Log_2021-12-01_16-33-41.log](https://github.com/OpenIntelWireless/itlwm/files/7632669/Log_2021-12-01_16-33-41.log)
Here is my sleep config, hope this useful.
[image]

zxystd, 2021-12-01:
@SesshomaruVN 
Your card's rfkill switch is persist closed after wakeup, the driver can't enable it.
[image]

SesshomaruVN, 2021-12-01:
> @SesshomaruVN
> Your card's rfkill switch is persist closed after wakeup, the driver can't enable it.
> [image]

is there any way I can fix it?

zxystd, 2021-12-01:
I tried forcing write the rfkill switch in driver now, although it should not do like this and should be managed by ACPI or BIOS.
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7632959/AirportItlwm.kext.zip)

SesshomaruVN, 2021-12-01:
> I tried forcing write the rfkill switch in driver now, although it should not do like this and should be managed by ACPI or BIOS.
> [AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7632959/AirportItlwm.kext.zip)

Unfortunate it is still not working. Thank for your help.
Maybe I should patch ACPI, but I don't know where to start. So could you recommend some useful knowledge that I can start from that. (I checked my bios setings and not relative to that one)
Thanks so much.

SesshomaruVN, 2021-12-02:
Hi guys, finally I found how to fix this problem.

If you are facing that one, you can try step by step - level of recommendation decreases.

1. You can try kext file that patched by @zxystd. I have found 2 version for [Bigsur](https://github.com/OpenIntelWireless/itlwm/files/7632046/Big.Sur.zip) and [Monterey](https://github.com/OpenIntelWireless/itlwm/files/7019815/Monterey.zip)
2. For the system as I described above, you need to turn off sleep mode. If you are using OC, you can try [their guide](https://dortania.github.io/OpenCore-Post-Install/universal/sleep.html) before. If not affect, you can use the command below in your terminal. ([Credit](https://gist.github.com/pwnsdx/2ae98341e7e5e64d32b734b871614915))
`sudo pmset -a sleep 0; sudo pmset -a hibernatemode 0; sudo pmset -a disablesleep 1;`
if you want to revert to default value, you can search in google or try this command:
`sudo pmset -a sleep 1; sudo pmset -a hibernatemode [original hibernatemode value]; sudo pmset -a disablesleep 0;`

Thanks @zxystd for helping me. Hope this guide can help more else guys.

Fuzail-maker, 2022-01-31:
@SesshomaruVN If you turn off sleep mode, doesn't it mean your laptop won't sleep?

Overc1ocker, 2022-02-21:
AC7265 Rev B, Monterey 12.2 sleep works properly with the latest 2.1.0 kext. Wifi instantly connects on wakeup.

williambj1, 2022-03-03:
@Overc1ocker Sounds more likely to be a motherboard flaw rather than a kext or wifi card issue. The card is dead locked by RFKILL after sleep.

usr-sse2, 2022-04-03:
For me, it doesn't work after sleep is `pci-aspm-default` is set to 0 (ASPM is disabled). If I remove this property (enable ASPM), then it works after sleep.

ferdysopian, 2022-11-13:
Don't know if it's the same on every motherboard, but on the "HP 14s dq1013tu" it has a Prepare to Sleep and Wake method that sets the WiFi and Bluetooth state.

I'm no expert on this but trying to figure it out.

First you need to verify what the WLBT device is trying to set on your DSDT.
[image]

Once you know what WLBT is trying to set up, check your MWAK and MPTS. There are some codes that are the same as those that appear on WLBT to save the current WiFi and Bluetooth status but for some reason, it's not working properly.
[image]

So you can rename MWAK and MPTS so they can be paired with a custom SSDT to force set WiFi and Bluetooth state on Wake.
[image]

With this configuration, WiFi and Bluetooth on my laptop will turn on when it wakes up.

closed 2024-03-16 by zxystd

VGerris, 2024-08-02:
> With this configuration, WiFi and Bluetooth on my laptop will turn on when it wakes up.

I have the exact same issue and the same code in BIOS.
I am stuck trying to compile the SSDT, can you post yours?
I just add a rename for MWAK to XWAK and then want to add the SSDT.aml for the last picture you sent.
Would be nice to get saving working but this is a good start, thanks a lot for sharing!

1alessandro1, 2025-01-11:
@VGerris here you go
[ssdt.zip](https://github.com/user-attachments/files/18386205/ssdt.zip)