OpenIntelWireless/itlwm #521: System unresponsive after playing music through airplay/用AirPlay播放音乐后系统未响应

wqhqq1, 2021-02-04:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.2.0
- WiFi Card Model: ax200
- Product ID: 0x8086(0x2723)
- macOS Version: 11.3(20E5172i)

**Description**

When I try to connect appletv and play music through airplay, system will be unresponsive until I disconnected appletv
**Bug Report Archive**

[image]

**Kext Download Source**

Github Release

win1010525, 2021-02-04:
Whole log please. Isn’t it a bug of Big Sur? Have you tried to restart your computer?

zxystd, 2021-02-04:
Nothing related to this driver.

closed 2021-02-04 by zxystd

wqhqq1, 2021-02-04:
你是说这是macOSbeta版的bug对吧

win1010525, 2021-02-04:
对，11.3bug特别多，我都服了

wqhqq1, 2021-02-04:
我换了好几个itwm还是如此，昨天作死升级的